opentx/opentx #3024: Companion 2.1 (windows) - Lua 'execute' fails in non-root directory

dsbeach, 2015-11-02:
Using the SD browser, select a .lua file, select 'Execute'.

Debug window reveals:
```
findTrueFileName(\SCRIPTS\ONETIME/play-dur.lua)
	not found
f_open(\SCRIPTS\ONETIME/play-dur.lua) = INVALID_NAME
Error in script \SCRIPTS\ONETIME/play-dur.lua: cannot open \SCRIPTS\ONETIME/play-dur.lua: Bad file descriptor
```

Apparently the companion doesn't like the mixed forward and backward slashes.

I'll create a pull request for the following addition to menu_general_manager.cpp line 241
```
#if defined(SIMU)
    for (int i = 0; lfn[i] != 0; i++) {
        if (lfn[i] == '\\') { lfn[i] = '/'; }
	}
#endif
```

bsongis, 2015-11-02:
In your commit message, would you write something like "Fixes #3024" so that the issue is automatically marked as fixed when it's merged into 'master'? Thanks!

dsbeach, 2015-11-02:
I did not know about that feature. Can do.

kilrah, 2015-11-03:
Are you sure that's a problem? There's always been a mix of forward and backslashes on Windows but it was never an issue before.

dsbeach, 2015-11-03:
It's definitely an issue on my MSVC build. I'll test against an official build today and add #if defined(MSVC) provided that is appropriate.

dsbeach, 2015-11-03:
Thanks for the link to the style guide, that will be useful.  I've adjusted my Visual Studio settings so at least the indentation and tabs -> spaces should be automatic from now on.

I reinstalled Companion 2.1.3 this morning and confirmed it is an issue in that release on my Windows 10 system.

Thanks for being patient, I think this is good to go.

kilrah, 2015-11-03:
> It's definitely an issue on my MSVC build. I'll test against an official build today and add #if defined(MSVC) provided that is appropriate.

Still wondering because I've been building and using on MSVC build since I started playing with OpenTX 3 years ago.
Not even talking of the officially released builds that are built the same way, and we never had a report of a related issue.

dsbeach, 2015-11-03:
I just confirmed the same issue on a Win 7 (32-bit and 64-bit).  Start the simulator, long press Menu, short press Page, navigate to SOUNDS, navigate to EN, navigate to a .wav file, press ENT, select PLAY.

Trace from debugger window:
```
f_opendir(.) = OK
f_opendir(.) = OK
f_opendir(.) = OK
f_getcwd() = \SOUNDS\en
playFile("\SOUNDS\en/acro.wav", flags=0, id=254)
findTrueFileName(\SOUNDS\en/acro.wav)
	not found
f_open(\SOUNDS\en/acro.wav) = INVALID_NAME
```

kilrah, 2015-11-04:
I had a look and as expected the problem is not what you think it is.
The mix of forward and backward slashes is not an issue, but when doing a "play" from the SD card manager the path the simulator searches in is incomplete, it omits to prepend it with the SD card root path configured in the profile (or at least a '.' tu use the current directory, but "\SOUNDS\en/acro.wav" is not a valid path on Windows (and wouldn't point where expected on Unix either).

From the SD browser we get, as you posted:

```
f_getcwd() = \SOUNDS\en
playFile("\SOUNDS\en/wizz.wav", flags=0, id=254)
findTrueFileName(\SOUNDS\en/wizz.wav)
	not found
f_open(\SOUNDS\en/wizz.wav) = INVALID_NAME
```

But if you play the sound in a custom function, no problem:
(selecting it)

```
findTrueFileName(D:\Electronique\9X\SD_taranis/SOUNDS/en/wizz.wav)
	found: wizz.wav
f_stat(D:\Electronique\9X\SD_taranis/SOUNDS/en/wizz.wav) = OK
f_opendir(D:\Electronique\9X\SD_taranis/SOUNDS/en) = OK
```

(triggering it)
```
playFile("/SOUNDS/en/wizz.wav", flags=0, id=1)
findTrueFileName(D:\Electronique\9X\SD_taranis/SOUNDS/en/wizz.wav)
	found in map: D:\Electronique\9X\SD_taranis/SOUNDS/en/wizz.wav
f_open(D:\Electronique\9X\SD_taranis/SOUNDS/en/wizz.wav) = 50F13088
f_close(50F13088)
```

projectkk2glider, 2015-11-04:
I messed with this part of the code and wrote `f_getcwd()`. I remember it was needed by the SD card browser. Apparently some part of the logic is broken, but the SD card browser relies on current working directory on the radio. Perhaps we need to convert the path to full path.

dsbeach, 2015-11-04:
Guys,

I'm sure there are several ways to address the issue, but I do want to point out that commit e48d58c does fix the issue. Note that the TRACE from playFile() is the same as when invoked from a custom function.

```
f_getcwd() = /SOUNDS/en
playFile("/SOUNDS/en/acro.wav", flags=0, id=254)
findTrueFileName(C:\Users\dbeach\Source\Workspaces\NemaSim\Solution\TrackerSim\emu_sdcard_2.1/SOUNDS/en/acro.wav)
	found: acro.wav
f_open(C:\Users\dbeach\Source\Workspaces\NemaSim\Solution\TrackerSim\emu_sdcard_2.1/SOUNDS/en/acro.wav, 1) = 508AF4F8
f_close(508AF4F8)
```

kilrah, 2015-11-04:
Your post just confirms what I said above. The slashes are not a problem, the path lacking the proper root is.

projectkk2glider, 2015-11-04:
Bug is in `convertSimuPath()` it searches only for `/` char, but is should also consider `\` char. It's a one-liner fix. @dsbeach will you do it?

Alternative fix is to change all `\` chars to `/` in `f_getcwd()` (in simpgmspace.cpp)

dsbeach, 2015-11-04:
Will do. I'll have a look tomorrow and update the pull request when it is tested.

commit 40e56573 (2015-11-19): Fixes #3024
commit a14e8a52 (2015-11-19): fixes #3024 - one line affected and a few parenthsis for clarity
commit ee95c973 (2015-11-19): Fixes #3024: Companion 2.1 (windows) - Lua 'execute' fails in non-root directory (ported from master, re #3025)
closed 2015-11-19 by dsbeach