opentx/opentx #3026: Companion Flight Modes, Logical Switches and Special Function Delete Action

elecpower, 2015-11-03:
Just a time saving feature with the addition of right mouse menu item to Delete/Reset the highlighted item to its defaults.

For Flight Mode it is the open FM tab and for LSs and SFs it is the highlighed row.

With us sharing model files or even repurposing our own there are times where there are many elements which are not required for the new model. Some will have no effect if old artefacts exist but others no so. Also results in cleaner model settings as we tweak over time. 

With ongoing development entries may become redundant as they are more easily handled by new features so you would like a quick tidy method.

Keep up the good work.

Taranis X9D Plus with v2.1.3

projectkk2glider, 2015-11-04:
I suggest we wait for #2861 to be finished before tackling this.

fraternl, 2015-11-05:
A great time saviour...
I had to completely redo a setup lately. I wish I had it then.

elecpower, 2015-11-19:
Also please add Telemetry Senors. When changing from Calculated to Custom and back to Calculated the Calculated row displays the old values.

projectkk2glider, 2015-11-22:
Well, the LS and SF already have a right click context menu (right click on `LS` label for example) where you can select Delete. So that is already done.

We need similar for:
* Flight mode
* Telemetry sensor (might be sensible to wait until #2781)
* Output

elecpower, 2015-11-23:
Appologies on not identying the existing context menus on LS and SF. I was over the first data field of each when right clicked rather than the labels. Great to see progress on the others.

closed 2020-07-07 by stale[bot]