opentx/opentx #3027: Changing play track to Override CH1 only hides the "enable field", it doesn't clear it

fraternl, 2015-11-05:
When I want to clear a special function I'm changing it to "Override CH1"  with a value of 0 and a "----" switch. 
If I do this with a "Play track" that has its "Enable" set to 60s this value is not shown anymore, but it's still there. 
I was doing a "model compare" and there this field showed up.

When I change it back to "Play track" this value appears again.
I don't know how far this hidden unused propery carries, but if it survives a back and forth from the radio this behaviour is unwanted.

It's a minor issue, but I think it would be better if that value is really cleared although it's not applicable for "Override CH"

kilrah, 2015-11-05:
Version? On companion or on the radio?

fraternl, 2015-11-05:
Today's nightly.... On Companion.

fraternl, 2015-11-05:
I just changed an LS from "a<x" to "Edge"
The behaviour is not the same.
"delay" is not applicable for Edge, but here this value is not hidden but greyed out.

I believe it's better to clear it.
Do you agree?

fraternl, 2016-01-13:
[image]
[image]

closed 2016-11-27 by fraternl