opentx/opentx #3028: Firmware 2.1.3 issue with OrangeRX module when switching between models.

indavis, 2015-11-05:
As reported on RCGroups by MrDuncan, he was having an issue when switching between models. His OrangeRX module(s) were not firing up when another model was picked. He went back to firmware 2.0.17 and stated it solved the problem. I requested that he post his 2.1.3 eepe up so that I could test with my Taranis and OrangeRX module. 

He posted it here:
http://www.rcgroups.com/forums/showthread.php?t=2451551&page=110

I loaded his eepe file and sure enough on firmware 2.1.3 I am experiencing the same issue. I will paste in what I found, because it seems that if you switch models and get a warning, either throttle or switch, it will re-initialize (or whatever happens, it beeps for me) your module. This is what I found and posted below his.

"Perfect. I loaded your eepe file into my radio and think I have found the sequence where the issue is occurring. 

Making sure the throttle is down, switching back and forth between models and the OrangeRX module stops beeping after switching a couple of times. On mine, it kept blinking, but the audible beep is not there.

Now here is where it gets odd. Since I have not adjusted my throttle (remind me to open up the radio and adjust that later). If I don't hold the throttle down I get the Throttle Warning, and I get the OrangeRX module beep every time. 

To test I confirmed switching back and forth between models 3 and 5 with the throttle held down (No throttle warning) and no confirmation beep from my OrangeRX module. The next time I switch to the other model and leave the throttle up and get the warning, after the warning goes away, we get the OrangeRX module beep. 

So best guess is that this is a code problem and it's somehow the Throttle Warning is bypassing the issue. Can anyone else on 2.1.3 with an OrangeRX module confirm this? I will see if I can put in a bug report in the meantime.

Edit: I just loaded my eepe back in and here is why I have never seen this. I have different models with differing switch configurations so it seems that each time I switch between models I get either the throttle or a switch warning, and after I clear the warning it resets the OrangeRX module. I copied a model so that I wouldn't get any warnings, and sure enough no beep on the module. "

It seems there is some sort of reproduceable issue with this. If you want me to attach his eepe I can.

kilrah, 2015-11-06:
Sounds simple, that module just doesn't like a quick power cycle.
Not really our problem since no other we know has issues with that...

griebd, 2015-11-06:
But if that is the case and there are a few users that suffers from this (OrangeRX are not that rare) we could do a simple workaround (in 2.2 as it would mean an eeprom change). An option that when checked in a model will produce a delay in powering the modules it uses.

bsongis, 2015-11-06:
Yes it sounds a good idea. Who wants to take it?

kilrah, 2015-11-06:
It could be always done, with no option. The delay probably only needs to be 100-200ms or so.

maddin21, 2016-05-28:
Hi, Just read abt that issue since today the same happened to me. I am using 2.1.8, when my Taranis plus is active with a model using the OrangeRX module  and I switch to another model (also configured to use the OrangeRX model) the receiver will not bind, the module seems to be switched off. Switching the Taranis off and then on again apparently resolves the problem. Most concerning is the fact that even everthing seemed to work, all of a sudden themodule switched off. Again, only switching off / on the Taranis resolved the problem for a while. I am not quite sure what exactly triggers that behaviour but I consider to switch back to 2.0.17 which was working without any issues. Will perform some tests before to figure out when exactly the problem occurs. Is there already a tentative date by when this timig thing will be fixed? thx :-)

closed 2020-07-07 by stale[bot]