opentx/opentx #3031: Set date and time not working

glini, 2015-11-06:
Radio 9RX Pro, opentx build 2.1.5; not able to set date and time. No tx-modul inserted, maybe it's because of this?

bsongis, 2015-11-06:
Was it ok in 2.1.4?

bsongis, 2015-11-06:
"Set menu long"? Are you in the radio setup page?

kilrah, 2015-11-07:
> Display says 'Set menu long'

Then you're on the wrong screen. Probably the VERSION screen, which is telling you the date/time the firmware was built and obviously can't be changed.

glini, 2015-11-07:
Radio setup page.  Right Column, Date-Time. In next page (Date-Time) I can change Sec., Min., Hour, Date, Month, Year. New settings reset when I leave page, long pressing Menu does not work.

kilrah, 2015-11-08:
Can you check with the original ersky9x firmware if it works well? Some have already had defective chips.

glini, 2015-11-08:
Checked with original ersky9x; working without problem.

glini, 2015-11-08:
Did a complete reinstall with companion. Seems this fixed the problem. Don't know what went wrong before, but now it works.

closed 2015-11-08 by glini

bsongis, 2015-11-08:
Ok, this is strange!