opentx/opentx #3032: Companion 2.1.5 on Ubuntu 10.15 Edit screen width

rdeanchurch, 2015-11-07:
The edit screen is much wider than it used to be in 2.0 and no longer fits on the screen of my laptop (Toshiba Satellite in  i386 mode ) ..  The width cannot be change smaller after switch from Setup.  Setup is slightly narrower and fits onto screen, but all the rest are very wide.

projectkk2glider, 2015-11-07:
My guess is that #2861will improve the situation a little bit, but perhaps the telemetry items need their own sub dialogs, like mixes.

griebd, 2015-11-14:
+1 to their own sub dialogs, like mixes.

elecpower, 2015-11-14:
+1 for sub dialogs

dk7xe, 2015-11-17:
I guess it's similar to the #2781 i have reported.

rdeanchurch, 2015-11-17:
Looks like exactly the same problem.  Your wording and description was superior to mine.
Dean

Calm down gravity is still working!! ....I just tested it.			 

Date: Tue, 17 Nov 2015 06:52:15 -0800

projectkk2glider, 2015-11-18:
Duplicate of #2781

closed 2015-11-18 by projectkk2glider