opentx/opentx #3034: [Companion Sim][Taranis B] Special Function AdjustGvar Bug ?

483917, 2015-11-07:
Wimdows 10 Pro Swedish
Companion 2.1.5N518 English
Taranis B, Firmware:
opentx-taranis-haptic-timer3-lua-mixersmon-sqt5font-en-2.1.5N519.bin

When using Special function , [AdjustGvar x] and lets say Rudder-trim as source
i have noticed occasionally that my rudder-trim value also changes, not ok i think

Take a look at my eeprom below and open model "AdjGVARtrim"
test in Companion simulator or on Taranis and set switch SE- to mid pos
start adjusting Gvar with Rudder-trim
and look at rudder-trim indicator , occasionally it changes

I tested and have the same behavior in Companion sim and Taranis B

https://www.dropbox.com/s/u6za2dt9pmg33wj/eeprom-160502-adjGvar-PlayCn.bin?dl=0

kilrah, 2015-11-07:
What you've put for L1 makes no sense, it's normal for it to behave as you describe as at times when you click the trim key the SF might be processed before L1 turns on. You just want to have SE- as the SF enable instead of L1.

closed 2015-11-07 by kilrah

483917, 2015-11-07:
Ok thanks :+1: