opentx/opentx #3037: FM0 switch assign

janekx, 2015-11-08:
Request from my customer
*****
Flight mode 0 has a possibility to assign a switch. After applying the switch, FM0 is enabled with its highest priority. This allows users to use FM0 in “panic”situations e.g. for safe landing.
(Currently FM1 needs to be utilized, which consumes a position of limited number of available Flight Modes). 
********
I can imagine that someone can have more FM then 8 so it can be useful.

kilrah, 2015-11-08:
Big change for something that can be done in many other ways - not for it at all. Firstly using FM1 whould not be a problem, then you could also gate the switches for other FMs with your panic switch using logical switches for example.

projectkk2glider, 2015-11-10:
One flight mode has to be the default - ie active when no other mode is active. This mode can not have a switch. The FM0 is now the default mode, hence it can't have a switch.

closed 2015-11-10 by projectkk2glider