opentx/opentx #3039: lua script less memory error since opentx 2.1.5

wolkstein, 2015-11-09:
i update my taranis today to 2.1.5.
now my telm1.lua stops with less memory error.

with 2.1.3 and the same setup i have no issues.

i have 4 mixer scripts running and one telem script.
this script are running. https://github.com/Clooney82/MavLink_FrSkySPort/tree/s-c-l-v-rc-opentx2.1/Lua_Telemetry/DisplayApmPosition/SCRIPTS
i have an extra MIXES script which not listed there.

my extra script
```lua
local FinaltPos = 0
local Incrementer = 0
local positionswitch = 0
local realdeathband = 0

local InSliderDeathBand = 0

-- Inputs HMI ----------------------------------
local InSpeed = 0		-- Input for desired global speed source device
local InSlider = 0		-- Input for desired position source device
local Deathband = 0             -- Input for desired deathband around input slider zero position

local function run ( InSpeed , InSlider, Deathband )
        
        realdeathband = ( 1024 / 100 ) * Deathband
        if InSpeed > 800 then
            positionswitch = 1
        elseif InSpeed < 800 and InSpeed >-800 then
            positionswitch = 2
        elseif InSpeed < 800 then
            positionswitch = 3
        end
        
        if InSlider >= 0 and InSlider <= realdeathband then -- zero position
            InSliderDeathBand = 0
        elseif InSlider > realdeathband then -- positiv values
            InSliderDeathBand = InSlider - realdeathband
        elseif InSlider <= 0 and InSlider >= realdeathband * -1 then -- zero negativ
            InSliderDeathBand = 0
        elseif InSlider <= realdeathband * -1 then --negativ valus
            InSliderDeathBand = InSlider + realdeathband
        end
        --print(InSliderDeathBand)
        
        Incrementer = InSliderDeathBand / ( 50 * positionswitch)
	-- End Compute final position
        -- print(InSliderDeathBand)
        FinaltPos = FinaltPos + Incrementer
        if FinaltPos >= 1024 then
            FinaltPos = 1024
        end
        if FinaltPos <= -1024 then
            FinaltPos = -1024
        end
	-- Function return results
	return FinaltPos 

end

return { run=run, input={{"Spd", SOURCE}, {"Sld", SOURCE}, {"Deat %", VALUE, 0, 100, 0}} , output={ "FPos"} } -- Return Functions to OpenTX
```

curios, if i remove all comments from all scripts and reduce the indent from 8 to 1 character my scripts run fine.
this is maybe an problem with the lua interpreter settings.

/g
wolke

projectkk2glider, 2015-11-09:
There are several possibilities:
* you have different firmware options selected
* the 2.1.5 has some new feature or fix that uses more memory
* we made a mistake somewhere and we unnecessary have less RAM for lua

I will check second and third possibility later.

wolkstein, 2015-11-09:
thx,
i also noticed that the the script continue to work if i change my timer0 and timer1 calls in script.

we need timer 1 and 2 to measure amed flight time and counting running hours from model.
if i change for example
``` lua
model.setTimer(1, {mode=0, start=0, value=0, countdownBeep=0, minuteBeep=false, persistent=1})
```
against
``` lua
model.setTimer(0, {mode=0, start=0, value=0, countdownBeep=0, minuteBeep=0, persistent=1})
```
the memory problem is not so hard and i can run the script.
the cons, if i use the model.setTimer Function in this way minuteBeep will ignored and i can hear an minute Timer call.

i am not sure if changing true/false against 1/0 can produce such a memory problem.

projectkk2glider, 2015-11-09:
Well the `0` in Lua actually means TRUE when evaluated a boolean! That explains why `minuteBeep=0` means the same as `minuteBeep=true`.

The `true/false` taking more memory than `0` might simply be a case of longer script source.

projectkk2glider, 2015-11-09:
The ram usage:
* 2.1.3 56096 bytes
* 2.1.5 56152 bytes

Difference is 56 bytes less available in 2.1.5 compared to 2.1.3. It either means that you were right on the limit and have now reached it or something else is broken. Judging by your comments that removing comments from the script fixes it I vote for first option.

I have checked why the 2.1.5 has bigger RAM usage and found nothing out of the ordinary, most of it can be attributed to a fix that corrected some variable alignment.

I will close this ticket now, if you have any other compelling reason feel fee to reopen it.

closed 2015-11-09 by projectkk2glider