opentx/opentx #3040: Companion Not All Settings Saved/Retrieved Between Sessions

elecpower, 2015-11-09:
The Application Setting 'Enable automatic backup before writing firmware' checked value is not saved across Companion sessions but it is within a session.

Steps to reproduce:
1. Start Companion
2. Open Settings
3. Select Application Settings tab
4. Enter or select a valid Automatic Backup Folder
5. Check Enable automatic backup before writing firmware
6. Click OK
7. Open Settings
8. Select Application Settings tab - checkbox is checked
9. Click OK or Cancel
10. Close Companion
11. Repeat steps 1 thru 3 - checkbox is unchecked.

Tested on Companion 2.1.5 Win 10 & Ubuntu 14.04

elecpower, 2015-11-24:
Let me prefix these comments with I have some programming experience in VB and COBOL but am a complete novice in C++ and zero with Git.
I followed the Wiki instructions for a Ubuntu build chain and downloaded  the master branch zip.
Bug appears to be the difference in the variable/setting name enableBackup and the ui dialog field name backupEnable. The inline definition macro assumes the names are identical so reverted to using the approach taken for other bool settings.
Made 4 changes to appdata.h and 3 to appdata.cpp, compiled and tested with success.
Would you like a copy of my changes and if so how?

projectkk2glider, 2015-11-24:
You can post an output of the `diff` command here or upload the whole new file somewhere and post link here.

elecpower, 2015-11-24:
Here are the diffs
[image]
[image]

projectkk2glider, 2015-12-26:
I say nothing is wrong here! The actual setting for backup is in application and profile settings. If you change it there the new value is saved. The check-box in firmware flash dialogue only enables you to change the pre-defined setting before each flash.

elecpower, 2015-12-26:
@projectkk2glider Just rechecked and the tick box value is saved from the radio profile tab but not from the application tab. Can you please reopen.

commit 4c113b65 (2015-12-26): Fixes #3040: Application Setting 'Enable automatic backup before writing firmware' was not saved
commit 86c5c312 (2015-12-26): Fixes #3040: Application Setting 'Enable automatic backup before writing firmware' was not saved (ported from master)
closed 2015-12-26 by projectkk2glider