opentx/opentx #3041: LUA telemetry script - different behavior between 2.0 and 2.1

Sacre100, 2015-11-09:
The following script has a different behavior in 2.0 and 2.1 : 
- 2.0 -> pressing PLUS or MINUS key, increase or decrease the value and **stay on the same screen** (OK for me)
- 2.1 -> pressing PLUS or MINUS key, increase or decrease the value but **switch to the next or previous screen** if any (not OK for me)

To test this in 2.1, define a telemetry screen attached to the script and **a second one** (for example one or more value). If you do not define at least a second telemetry screen you will not see the problem.

Here is the script (only code which show the issue) : 

      local idx = 0
      return {
        run = function(event)
          if event == EVT_PLUS_FIRST then
            idx = idx + 1
            killEvents(EVT_PLUS_FIRST)
            killEvents(EVT_PLUS_BREAK)
          elseif event == EVT_MINUS_FIRST then
            idx = (idx - 1)
            killEvents(EVT_MINUS_BREAK)
            killEvents(EVT_MINUS_FIRST)
          end
          lcd.clear()
          lcd.drawText(10, 20, "idx : ", 0)
          lcd.drawText(lcd.getLastPos(), 20, idx, 0)
        end
      }

projectkk2glider, 2015-11-09:
Could we mark this a as duplicate of #2469?

Sacre100, 2015-11-09:
Yes, I search on "+", "-", LUA, but I didn't search on "killEvents", "plus", "minus", sorry, I close this one.

closed 2015-11-09 by Sacre100