opentx/opentx #3042: [Companion Sim][Taranis B] Special Function->PlayValue->Channel(x) dont play actual Output channel(x) value

483917, 2015-11-09:
Windows 10 pro swedish
Companion 2.1.5N521
Taranis B Firmware:
opentx-taranis-haptic-timer3-lua-mixersmon-sqt5font-en-2.1.5N521.bin

I noticed that if i use either a Special Function, OverrideCH(x) 
or using a Curve on Output(x) "Servo output page"
the playValue Channel(x) function plays the wrong value ?

It plays the "Mixer output value" and not the actual "Output value"

If this is intended behavior ?,then feel free to close this issue
But feels confusing

Open my demo file:
https://www.dropbox.com/s/6uon7t6796taor0/Special%20Function%20playValue.eepe?dl=0

And open Model , OverrideCHx , use switch SF to enable OverrideCH3 on/off
toggle switch SH to play value while moving Gas-Stick and look at channel 3

and/or open Model , OutputCurve ,move Gas-Stick and toggle switchSH while looking at
channel 3

kilrah, 2015-11-09:
Yes it's normal. Leaving it open for discussion whether it's a good thing or could warrant a change, but it's always been like that.

projectkk2glider, 2015-11-10:
This function should in my opinion play the same value as is displayed in Channel monitor screen.

kilrah, 2015-11-10:
I'd vote for that too.

bsongis, 2015-11-10:
Ok

projectkk2glider, 2015-12-13:
This function uses getValue(<channel>) to get the current channel value. The getValue() returns values from ex_chans[] (raw mixer outputs before limits and failsafe). 

I think the getValue() should be changed to return the values from channelOutputs[] (after limits and failsafe).

But this change will also affect other parts of the code where getValue() for channel is used. I looked where this is actually used and these are the areas:
* Horus curves
* Telemetry screens (show channel value)
* logical switches

The most problematic is a logical switch that uses channel value, with this change the value could be different (if user set output limits to non-default values).

Another option is to add a whole new set of mixer sources - **channel outputs** that would actually return a final channel value. 

Another simpler option is to only rewrite the play value function to use actual output values, but this will make the CHx reference inconsistent (possibly different value in play value and logical switches)

I am not sure what to do! @bsongis @kilrah

Spainman, 2016-05-10:
First of all I'm not sure what others may use this functionality for.... 
My use case typically is to announce the percentage of certain secondary controls like flaps, camber, governor rpm preset, gyro gain. Obviously I want to hear the percentage (= mixer output) of e.g. positive/negative deployment of the control surfaces according to the planes logical configuration. For regular flaps I even go so far to add a mixer line that maps the -100, 100 range to a 0-40° range just for a proper voice announcement.

The channel output value that goes to the servos after going throw the outputs stage (throw calibration, curve tweaks like linearization and what not) is of absolutely no value for an in-flight announcement - I'm not the servo, I'm the pilot! 

Please don't break it by changing the behavior just to conform with the lettering of the functions name. If you really feel that there is a need to read out the real channel output value, then please at least provide an alternative function that still allows to announce the mixer output without having to do extra workarounds.

kilrah, 2016-05-11:
Yeah the only option is to rename the current sources as "Mixer outputs" and add new "Channel outputs" sources.

483917, 2016-05-11:
I agree fully, 
Rename the current sources as "Mixer outputs" and **add new** "Channel outputs" sources

3djc, 2016-05-23:
I have a similar need for "true" output reading, I would like to sugest however that ch1 to ch32 stay untouched (not renamed), to keep most things working. out1 to out32 could be created, that would leave flexilbility to change (or not) the source of functions on a case by case.basis

lshems, 2016-05-23:
In my opinion this is only part of the solution.

I think with the introduction of global functions, a separation between mixer output and servo input is missing. What people intend to do with global functions is to standardize some use cases, as mentioned above.

Well, how do you deal when suddenly you need another servo layout? You have to reorganise your mixrrs to match them, and loose the global function mapping.

Opentx is a framework that supports technical mapping to achieve anything you like. Once you start creating functional layers, you are bound to run into problems.

I only use lua to do the functional part, even as a GUI for model adjustments, but all from a functional point of view. The technical stuff is in opentx.

I never use the global functions. I don't need them, because I need another set for ever model setup. I use my lya to make sure all models function the same on a gui level.

kilrah, 2016-05-23:
Completely off topic...

projectkk2glider, 2016-05-23:
@lshems #3263

kilrah, 2016-05-24:
But this issue is neither about global vars nor about adding a layer... It's about giving access to an extra thing that's already there.

lshems, 2016-05-24:
Never mind. Global functions will be a pain ita and cause a lot of confusion/extra work. But ok, they are there, so ...

I really hope it's not going to complicated things to much.