opentx/opentx #3043: Aspd bad values

janekx, 2015-11-09:
I think we got in trouble with Aspd measure. Tested 2.1.3 and 2.1.5 public

Used ASS-70 and ASS-100 both same issue. 

Actual / measured in kmh 
30 / 12
50 / 21
80 / 33
100 / 42

Actual / mesured in mph
30 / 19
50 / 32
60 / 40

From this I see that there is some calculation issue. In fact I cant keep car running exact same speed there can be 5-10% fluctuation, but this is too much :) 

in kmh there is 2.5 multilier to got actual speed 
in mph is 1.6 multiplier

When we not use the filter we got the bas screens values as well, the Aspd+ and - stop changing and show absolute bad values with bad signs, here is the photo, when we used filter never come back. 

[image]

projectkk2glider, 2015-11-10:
The manual for ASS is very bad, but it mentions a "Reset button" and it supposedly does a "user calibration". @janekx do you know how the user calibration works?

And just to confirm what you are saying above: when the car was going 50 km/h you got 32 mph and  21 km/h readings on the radio (depending on the unit that is setup for the sensor).

janekx, 2015-11-10:
Yes I know calibration procedure, it is simble reset, no wind and pressing will zero the device. 
I make many tries. With switched Aspd senzor to kmh and next switched to mph. I switch car  speedmeter to kmh or mph to see the speed as well so table kmh is all in kmh and mph as all in mph.

griebd, 2015-11-10:
just a question, should a air speed work as expected near ground?! you mentioned a car...
never looked at air flow in a car... but unless both air inputs are really well placed I wouldn't expect it to have correct measures.

janekx, 2015-11-10:
Please be sure i not measure my air-conditioning

projectkk2glider, 2015-11-10:
What griebd means that air speed sensor should be in **clean air**. In your case you should put it on a long pole and stick it out of the car at least 1 meter if not more.

janekx, 2015-11-10:
about 0.7m off the car diagonaly up and right. There is no much air mass around car in lower speeds, it grows how does car goes faster. 
I am sure we make the test well, I was driving and colegue was holding the rod off the car.

janekx, 2015-11-10:
By the way the test was done based on the client info that it measure too low, because we were not able to measure the plane well, we choose car as some kind of reference.

projectkk2glider, 2015-11-10:
@janekx do you have a log file of the testing, it might give us some additional clues.

janekx, 2015-11-10:
No, I have logged but forget to check at sensor log option :( I can make the log, we spend about hour to measure it. It measure well but wrong values.

projectkk2glider, 2015-11-11:
Some findings:
* OpenTX code is missing conversions where the source unit is UNIT_KMH which means that setting different unit on air speed sensor has no effect (confirmed in Companion simulator).
* all int32_t sensor values get interpreted by the OpenTX correctly  (confirmed in Companion simulator).
* none of developers had an air speed sensor to test, I also remember that specification for the sensor changed (unit from knots to km/h).
* based on @janekx the data from the sensor looks very noisy. Is there more that just the speed sent in each packet? Perhaps something else is packed into 4 bytes of data? That would explain the wrong readings as well as noisy signal (large max value).

janekx, 2015-11-11:
Yes it is good idea,  i can test all ASS types and N2 as well have here.

projectkk2glider, 2015-11-11:
@janekx for which radio should we prepare the custom firmware?  X9E, Taranis or Taranis Plus?

janekx, 2015-11-11:
X9E

janekx, 2015-11-11:
I was looking around for certified blower, my wifes hair drier have no speed certificate and all other fans I have here are super silend so slow... 
Seriously... 
two speed hair drier make big fluctuation in speed from 17-25 and 25-30 units. 
fans are too slow about 4-8 units  
car was much better, speed was nice stable and we see what speed we ride.

projectkk2glider, 2015-11-11:
No need for super accurate speed reading now, we just want to see the data stream at several speeds.

Special OpenTX version with S-PORT logging for X9E is available here: http://www.filedropper.com/opentx-215-x9e-sportlog

Instructions:
* backup model settings if they are important to you before flashing
* do not use this firmware for flying!
* every time you start the radio with this firmware a file `sport.log` will be created in the root directory of SD card. This file will contain a raw log of all received S-PORT data. We are interested in the contents of this file. Expect the file contents to grow quickly, a couple of mega bytes is nothing unusual.
* The sensor does not have to be detected for the log file to work, but it does not hurt and detecting sensors should help you see if the sensor is sending any data.
* After each test session make a copy of the `sport.log` file and rename it to something meaningful. Then delete the file on SD card. This way each session (possibly with different sensors) will have its own file.
* Upload files somewhere and post links here with a short description of what each file represents (which sensor was connected, what did you do, etc..)
* To test the air speed sensor, first leave it alone (speed should be close to zero, maybe even calibrate it) and then blow in it a couple of times so we can get the speed data.

Thanks for you effort.

janekx, 2015-11-11:
Here you have it http://rcstudio.cz/LOGS.zip
There is each test zipped to archive that tell what was done and what sensor was used.

projectkk2glider, 2015-11-11:
@janekx I analyzed your data, it all seems perfectly OK, no anomalies. How did the Aspd- and Aspd+ behave in this tests with this special firmware? I can not say nothing about the speed accuracy, but otherwise the data is as expected.

Examples of parsed packets:
```
packet: e9:10:00:0a:00:00:00:00:e5 (  45)  Aspd: 0.0 m/s
packet: e9:10:00:0a:14:00:00:00:d1 (  47)  Aspd: 2.0 m/s
packet: e9:10:00:0a:1e:00:00:00:c7 (  49)  Aspd: 3.0 m/s
packet: e9:10:00:0a:1e:00:00:00:c7 (  50)  Aspd: 3.0 m/s
packet: e9:10:00:0a:b4:00:00:00:31 ( 593)  Aspd: 18.0 m/s
packet: e9:10:00:0a:37:01:00:00:ad (2733)  Aspd: 31.1 m/s
packet: e9:10:00:0a:38:01:00:00:ac (2737)  Aspd: 31.2 m/s
packet: e9:10:00:0a:38:01:00:00:ac (2743)  Aspd: 31.2 m/s

janekx, 2015-11-11:
I think I had filter enabled that stay from setting before.  No issues about + and -  but what about final displayed speed?

projectkk2glider, 2015-11-11:
You had `Auto offset` checked for the ASS100 air speed sensor! I think it should not be checked, otherwise the speed will be wrong!

janekx, 2015-11-11:
Yes auto offset and filter but the values was same as we tested in car day before. So we test filter because strange numbers and auto offset. 
You probable got good data but display bad ones

janekx, 2015-11-11:
OK again, 
This is ASS100 auto offser off, filter off, kmh 

Max speed I saw is 25 kmh 

what you will see from raw data ?
[RCStudio-2015-11-15.txt](https://github.com/opentx/opentx/files/32414/RCStudio-2015-11-15.txt)
[sport.txt](https://github.com/opentx/opentx/files/32415/sport.txt)

projectkk2glider, 2015-11-11:
So far I am not seeing any problem with the air speed measurement with the custom firmware. @janekx  would it be possible to test the sensor in the car again with the custom firmware? The log files might tell us something at higher air speeds.

Your sport.txt gives this (looks perfectly ok):
```
packet: e9:10:00:0a:14:00:00:00:d1 (  19)  Aspd: 2.0 m/s
packet: e9:10:00:0a:14:00:00:00:d1 (  19)  Aspd: 2.0 m/s
packet: e9:10:00:0a:14:00:00:00:d1 (  19)  Aspd: 2.0 m/s
packet: e9:10:00:0a:1e:00:00:00:c7 (  20)  Aspd: 3.0 m/s
packet: e9:10:00:0a:1e:00:00:00:c7 (  21)  Aspd: 3.0 m/s
packet: e9:10:00:0a:1e:00:00:00:c7 (  24)  Aspd: 3.0 m/s
packet: e9:10:00:0a:1e:00:00:00:c7 (  26)  Aspd: 3.0 m/s
packet: e9:10:00:0a:1e:00:00:00:c7 (  32)  Aspd: 3.0 m/s
packet: e9:10:00:0a:1e:00:00:00:c7 (  33)  Aspd: 3.0 m/s
packet: e9:10:00:0a:14:00:00:00:d1 (  34)  Aspd: 2.0 m/s
packet: e9:10:00:0a:1e:00:00:00:c7 (  35)  Aspd: 3.0 m/s
packet: e9:10:00:0a:14:00:00:00:d1 (  38)  Aspd: 2.0 m/s
packet: e9:10:00:0a:14:00:00:00:d1 (  39)  Aspd: 2.0 m/s
packet: e9:10:00:0a:1e:00:00:00:c7 (  42)  Aspd: 3.0 m/s
packet: e9:10:00:0a:14:00:00:00:d1 (  44)  Aspd: 2.0 m/s
packet: e9:10:00:0a:1e:00:00:00:c7 (  46)  Aspd: 3.0 m/s
packet: e9:10:00:0a:14:00:00:00:d1 (  48)  Aspd: 2.0 m/s
packet: e9:10:00:0a:14:00:00:00:d1 (  50)  Aspd: 2.0 m/s
packet: e9:10:00:0a:14:00:00:00:d1 (  51)  Aspd: 2.0 m/s
packet: e9:10:00:0a:14:00:00:00:d1 (  55)  Aspd: 2.0 m/s
packet: e9:10:00:0a:1e:00:00:00:c7 (  56)  Aspd: 3.0 m/s
packet: e9:10:00:0a:1e:00:00:00:c7 (  59)  Aspd: 3.0 m/s
packet: e9:10:00:0a:28:00:00:00:bd (  61)  Aspd: 4.0 m/s
packet: e9:10:00:0a:28:00:00:00:bd (  63)  Aspd: 4.0 m/s
packet: e9:10:00:0a:14:00:00:00:d1 (  65)  Aspd: 2.0 m/s
packet: e9:10:00:0a:14:00:00:00:d1 (  68)  Aspd: 2.0 m/s
packet: e9:10:00:0a:28:00:00:00:bd (  69)  Aspd: 4.0 m/s
packet: e9:10:00:0a:1e:00:00:00:c7 (  74)  Aspd: 3.0 m/s
packet: e9:10:00:0a:1e:00:00:00:c7 (  75)  Aspd: 3.0 m/s
packet: e9:10:00:0a:14:00:00:00:d1 (  89)  Aspd: 2.0 m/s
packet: e9:10:00:0a:1e:00:00:00:c7 ( 117)  Aspd: 3.0 m/s
packet: e9:10:00:0a:1e:00:00:00:c7 ( 146)  Aspd: 3.0 m/s
packet: e9:10:00:0a:1e:00:00:00:c7 ( 180)  Aspd: 3.0 m/s
packet: e9:10:00:0a:1e:00:00:00:c7 ( 206)  Aspd: 3.0 m/s
packet: e9:10:00:0a:28:00:00:00:bd ( 230)  Aspd: 4.0 m/s
packet: e9:10:00:0a:28:00:00:00:bd ( 254)  Aspd: 4.0 m/s
packet: e9:10:00:0a:1e:00:00:00:c7 ( 284)  Aspd: 3.0 m/s
packet: e9:10:00:0a:28:00:00:00:bd ( 311)  Aspd: 4.0 m/s
packet: e9:10:00:0a:28:00:00:00:bd ( 331)  Aspd: 4.0 m/s
packet: e9:10:00:0a:28:00:00:00:bd ( 361)  Aspd: 4.0 m/s
packet: e9:10:00:0a:1e:00:00:00:c7 ( 386)  Aspd: 3.0 m/s
packet: e9:10:00:0a:1e:00:00:00:c7 ( 410)  Aspd: 3.0 m/s
packet: e9:10:00:0a:28:00:00:00:bd ( 440)  Aspd: 4.0 m/s
packet: e9:10:00:0a:28:00:00:00:bd ( 465)  Aspd: 4.0 m/s
packet: e9:10:00:0a:28:00:00:00:bd ( 487)  Aspd: 4.0 m/s
packet: e9:10:00:0a:1e:00:00:00:c7 ( 511)  Aspd: 3.0 m/s
packet: e9:10:00:0a:28:00:00:00:bd ( 541)  Aspd: 4.0 m/s
packet: e9:10:00:0a:1e:00:00:00:c7 ( 567)  Aspd: 3.0 m/s
packet: e9:10:00:0a:1e:00:00:00:c7 ( 590)  Aspd: 3.0 m/s
packet: e9:10:00:0a:1e:00:00:00:c7 ( 621)  Aspd: 3.0 m/s
packet: e9:10:00:0a:28:00:00:00:bd ( 646)  Aspd: 4.0 m/s
packet: e9:10:00:0a:28:00:00:00:bd ( 669)  Aspd: 4.0 m/s
packet: e9:10:00:0a:28:00:00:00:bd ( 700)  Aspd: 4.0 m/s
packet: e9:10:00:0a:1e:00:00:00:c7 ( 725)  Aspd: 3.0 m/s
packet: e9:10:00:0a:28:00:00:00:bd ( 749)  Aspd: 4.0 m/s
packet: e9:10:00:0a:1e:00:00:00:c7 ( 780)  Aspd: 3.0 m/s
packet: e9:10:00:0a:28:00:00:00:bd ( 805)  Aspd: 4.0 m/s
packet: e9:10:00:0a:1e:00:00:00:c7 ( 830)  Aspd: 3.0 m/s
packet: e9:10:00:0a:1e:00:00:00:c7 ( 853)  Aspd: 3.0 m/s
packet: e9:10:00:0a:28:00:00:00:bd ( 884)  Aspd: 4.0 m/s
packet: e9:10:00:0a:28:00:00:00:bd ( 909)  Aspd: 4.0 m/s
packet: e9:10:00:0a:1e:00:00:00:c7 ( 934)  Aspd: 3.0 m/s
packet: e9:10:00:0a:28:00:00:00:bd ( 964)  Aspd: 4.0 m/s
packet: e9:10:00:0a:28:00:00:00:bd ( 989)  Aspd: 4.0 m/s
packet: e9:10:00:0a:1e:00:00:00:c7 (1014)  Aspd: 3.0 m/s
packet: e9:10:00:0a:1e:00:00:00:c7 (1045)  Aspd: 3.0 m/s
[... 292 lines cut ...]
packet: e9:10:00:0a:5a:00:00:00:8b (8997)  Aspd: 9.0 m/s
packet: e9:10:00:0a:5a:00:00:00:8b (9027)  Aspd: 9.0 m/s
packet: e9:10:00:0a:50:00:00:00:95 (9052)  Aspd: 8.0 m/s
packet: e9:10:00:0a:5a:00:00:00:8b (9075)  Aspd: 9.0 m/s
packet: e9:10:00:0a:5a:00:00:00:8b (9110)  Aspd: 9.0 m/s
packet: e9:10:00:0a:5a:00:00:00:8b (9131)  Aspd: 9.0 m/s
packet: e9:10:00:0a:5a:00:00:00:8b (9158)  Aspd: 9.0 m/s
packet: e9:10:00:0a:5a:00:00:00:8b (9190)  Aspd: 9.0 m/s
packet: e9:10:00:0a:50:00:00:00:95 (9214)  Aspd: 8.0 m/s
packet: e9:10:00:0a:64:00:00:00:81 (9238)  Aspd: 10.0 m/s
packet: e9:10:00:0a:64:00:00:00:81 (9269)  Aspd: 10.0 m/s
packet: e9:10:00:0a:5a:00:00:00:8b (9292)  Aspd: 9.0 m/s
packet: e9:10:00:0a:5a:00:00:00:8b (9318)  Aspd: 9.0 m/s
packet: e9:10:00:0a:5a:00:00:00:8b (9344)  Aspd: 9.0 m/s
packet: e9:10:00:0a:5a:00:00:00:8b (9376)  Aspd: 9.0 m/s
packet: e9:10:00:0a:5a:00:00:00:8b (9399)  Aspd: 9.0 m/s
packet: e9:10:00:0a:5a:00:00:00:8b (9424)  Aspd: 9.0 m/s
packet: e9:10:00:0a:5a:00:00:00:8b (9454)  Aspd: 9.0 m/s
packet: e9:10:00:0a:64:00:00:00:81 (9477)  Aspd: 10.0 m/s
packet: e9:10:00:0a:64:00:00:00:81 (9503)  Aspd: 10.0 m/s
packet: e9:10:00:0a:5a:00:00:00:8b (9535)  Aspd: 9.0 m/s
packet: e9:10:00:0a:64:00:00:00:81 (9560)  Aspd: 10.0 m/s
packet: e9:10:00:0a:5a:00:00:00:8b (9584)  Aspd: 9.0 m/s
packet: e9:10:00:0a:64:00:00:00:81 (9609)  Aspd: 10.0 m/s
```

janekx, 2015-11-11:
So you not calculate the datas from m/s to kmh. On smartport protocol I see (km/10)/Hour

kilrah, 2015-11-12:
Unit according to the protocol doc I have is indeed km/h/10. That's correct in OpenTX, UNIT_KMH, PREC1. 

What's missing is the unit conversions, currently for speed the only conversions that are run are from knots to the others. 
https://github.com/opentx/opentx/blob/master/radio/src/telemetry/telemetry.cpp#L550

You definitely do NOT want auto offset on for an airspeed sensor.

projectkk2glider, 2015-11-12:
Added issue #3059 for the conversions from km/h

janekx, 2015-11-12:
I will make the test with ASS-100 in car and log it, if the displayed values are same like values in raw data log, it is not kmh speed and not km/h/10 too. 
As I write the displayed speed needs to multiply by 2.5 to get close to real car speed.

janekx, 2015-11-12:
So here is test in car, auto offser off, filter off, kmh
[sport.txt](https://github.com/opentx/opentx/files/33246/sport.txt)
[RCStudio-2015-11-16.txt](https://github.com/opentx/opentx/files/33247/RCStudio-2015-11-16.txt)

Video will be here soon https://youtu.be/Z2O25-giTTY

projectkk2glider, 2015-11-12:
@janekx Your last data set looks perfectly OK. The raw values from sport.txt are parsed correctly. There are no anomalies. 

Based on your reports and the video we can see that the measured speed in km/h and actual speed of the car differ a lot (about a factor or 2.5). But all I can say that this is not an OpenTX bug. The fact is that sensor sends such values. 

So we should contact FrSky (I bet you can do that) and ask them why there is such difference in measured and actual speed. Perhaps sensor sends speed in some other unit or has some other problem. If you have the time and the will you could do further tests:
* do a car measurement at more speeds to see if each time the error factor (2.5) is the same. That would at least prove that there is some kind of scale/unit problem and not something else.
* do a test with another FrSky speed sensor to compare if it has the same error factor as the first one. You already seem to have tested the N2 sensor? Did that one give more accurate result?

janekx, 2015-11-12:
Did not I tell  it at first article? Why I must make these test again? I am untrustworthy? My car method finally looks best what we can do....
Only one question, what you think will happend when I connect this setup to T+ with 2.0.16 OpneTX? 
Hint: we not need FrSky, there  is bug in the OpentX 2.1.X

bsongis, 2015-11-12:
I think @projectkk2glider would like to know if the difference is linear (I mean the same ratio at slow speed and at high speed).
Was it ok in 2.0.X?

projectkk2glider, 2015-11-12:
Your first data set looks strange. The measured speed should be the same regardless of the unit selected, because the unit conversion is not implemented yet.

You don't have to do the tests if you don't won't. I wrote that already. I have no personal desire to have the air speed sensor working at the present time, I don't even own one and probably never will. I am trying to solve your problem.

 The 2.0.x will treat sent speed in knots, but that is a factor of 1.8 not 2.5.

janekx, 2015-11-12:
First article are measured tables that show all... 
And the winner Bertrand,  yes 2.0.16 shows values 46 when 2.1 shows 20,  so when we count some %error it shows well

janekx, 2015-11-12:
Where my data looks strange? 
One table is in kmh and second is in mph 30mph is 50kmh and measured is 19 and 21 it is ok

projectkk2glider, 2015-11-12:
So 50 in your second table is actually 50 mph? That would explain the difference. I thought it was 50 km/h.

janekx, 2015-11-12:
Yes second is all in mph first is all in kmh I told that I switch speedmeter of the car to kmh and then to mph

bsongis, 2015-11-12:
I am really afraid that OpenTX is right here, the S.PORT protocol shows 0x000000e6, which means 23.0km/h, not 50km/h

We need to ask @Adelaxie

projectkk2glider, 2015-12-13:
As far as I can reconstruct from the emailing from FrSky the unit that sensor uses is actually **knots**. Do you agree? We just need to change the default unit and this will be done.

janekx, 2015-12-13:
Yes!  As i wrote in mails the windmaster prove that it is knots.

projectkk2glider, 2015-12-13:
Ok, I will fix this tonight.

dsbeach, 2015-12-14:
With this change I'll need to make changes in the telemetry simulator log file playback at a minumum.  Are you OK with having the sim display and accept values in km/h which we be converted to knots when sent to the transmitter?  That is the way GSpd is being handled now.

projectkk2glider, 2015-12-14:
Both ways are OK with me as long as both (GPS and Aspd) are done it the same way.

dsbeach, 2015-12-14:
Good - I'll get a pulls done for both master and next today.

kcaldwel, 2016-02-11:
I just measured values with an N2 airspeed sensor and an OXS GPS on a car (about 1m in front and above the car) that are about 2x the actual values for both the airspeed and the GPS ground speed.

I am using OpenTx 2.1.7,

I had kmh set for units on both Gspd and Aspd. My Taranis B model was set to metric.

I believe I was getting correct speeds from a FrSky GPS in an earlier version of OpenTx 2.1+, but the N2 airspeed readings were incorrect then because I didn't have the 47.1 ratio.

Should I not use the 47.1 correction factor in 2.1.7? Is the GPS speed problem with OXS or OpenTx?

Thanks!

nigelsheffield, 2016-02-12:
GPS speed sent by sensor is in knots isn't it?
At least it's whats expected in 2.0 so sensor will still be sending knots regardless of what the  tx operating system is won't it?

projectkk2glider, 2016-02-12:
@kcaldwel There was a change in the unit. Originally the FrSky specification had km/h for the Air speed unit. After the testing we found out that it was wrong and that the actual unit is knots. So we changed it in 2.1.7.

kcaldwel, 2016-02-12:
If I understand correctly, I no longer need the 41.7 ratio for the N2 airspeed sensor.

And I need to re-flash my OXS to send knots, not kmh.

Thanks!

kilrah, 2016-03-02:
> There was a change in the unit. Originally the FrSky specification had km/h for the Air speed unit. After the testing we found out that it was wrong and that the actual unit is knots. So we changed it in 2.1.7. 

Reopening this as following the RCG discussion yesterday I asked Adela again... and FrSky assure me 100% that their airspeed sensor supplies km/h, not knots. 

Whoever did the measurement could have forgotten that the radio was already applying a conversion...

janekx, 2016-03-02:
This look like immortal bug. Can you please ad the link on discussion?

kilrah, 2016-03-02:
http://www.rcgroups.com/forums/showthread.php?t=2178865&page=692#10376

kcaldwel, 2016-03-04:
I just ran a quick first try with my manometer airspeed calibration set-up and the FrSky ASS-70 using OpenTX 2.1.7.. It is a windy day, and the static pressure in the building was fluctuating. That makes the lower speed readings a bit unreliable, but as a test I think it worked OK. I have a couple of improvements to make to get consistent readings as well.

I had the Tx telemetry set to knots. I believe the FrSky -70 is sending knots, and reads about 5% low across the range. I will hopefully test the N2 airspeed sensor soon.

[image]

kcaldwel, 2016-03-05:
I got a quick run in on the N2 airspeed. Again, we have a big windstorm at the moment which is making the static pressure in the building unreliable, so the results aren't as solid as they should be. I am using Richard Johnson's (full-size sailplane test guru) method of calibrating airspeed against a known pressure in a manometer. It should be accurate, but a calm day would help! Tghe pressure changes are very small, especially at the lower airspeed values.

The N2 sensor I have has an error in slope the other direction from the FrSky. It sends one tenth knots, while the ASS-70 only reads whole knots. There could be quite a bit of variation in the individual sensor response to pressure, so it is quite possible other sensor of the same make will show completely different errors. I haven't checked the spec sheets for the +/- to see if these results are within the sensor variations.

[image]

projectkk2glider, 2016-03-05:
The sensor sample count is low so nothing substantial can be concluded yet, but it looks like both sensors are reporting speed in knots as predicted and measured earlier by @janekx The code is already up to date with that, only FrSky's SPORT specification is still wrong.

janekx, 2016-03-05:
@projectkk2glider yes I think so. There was detailed research in this area and direct communication with FrSky agreed the knots before and tests proove it

kilrah, 2016-03-05:
The thing is their doc still mentions km/h, and when I asked them last week I was told it was "km/h for sure"...

janekx, 2016-03-05:
Yes,  but we discuss it in mails with Adela and other engeneer and was confirmed that kmh are wrong in protocol datasheet. 
As well we make measurement with windmaster2 anemometer and it confirm knots.

kilrah, 2016-03-05:
Well again Adela and their engineer just told me it was definitely km/h... which is why I reopened so that we recheck to be sure. Looks like knots are right indeed.

kcaldwel, 2016-03-05:
The manometer method I am using is the same method they use to calibrate full-size aircraft airspeed indicators with, except I am using distilled water (density corrected for temperature) instead of mercury to increase the low end sensitivity as Richard Johnson did in his set-up. It is calibrated to the ICAO Standard Atmosphere conditions. It should be the best method to actually calibrate an airspeed sensor.

I still have a lot of wind here today, so there is no point in testing. I will have an oXs sensor ready to test shortly.

There is no way the FrSky ASS-70 is sending kmh, it is even 5% low for knots. 5% error for the low precision version of an uncalibrated sensor is reasonably good. The N2 is closer to sending mph than knots: 1.15mph/knot; the N2 is sending 12% high.

projectkk2glider, 2016-04-29:
@kilrah any progress? I think we can close this issue.

janekx, 2016-04-29:
I got from FrSky some custom ASS firmware but finaly we switch back  to factory and it is ok

commit 57e3d1af (2015-12-13): Fixes #3043: unit for Air speed sensor set to knots
commit aa2ca4f7 (2015-12-13): Fixes #3043: unit for Air speed sensor set to knots