opentx/opentx #3047: Alt in top bar

PieroCo, 2015-11-10:
Taranis x9d and opentx 2.1.5

In the Companion Telemetry Top Bar when I select Alt, (vario) as the Altitude source all seems OK on the Taranis initially. Observing the Top Bar on the Taranis it will randomly display ft, (x feet) or a number. The number will be 4294967xx, (xx will vari 20-29) and if the vario tone is on, a unusual tone is produced at the same time the 4294967xx is displayed. Sometimes the number continues to be displayed, (constant with xx variable) but vario tone appears to return to normal.

projectkk2glider, 2015-11-10:
What are your altitude sensor and receiver?

PieroCo, 2015-11-10:
Frsky Variometer Sensor-high precision and X6R

projectkk2glider, 2015-11-10:
Was it working in any previous 2.1.x release?

projectkk2glider, 2015-11-10:
Might be related to #3043 which also displays unusually large numbers.

PieroCo, 2015-11-10:
Same problem with d4rii but Ok with x8r

I updated directly from 2.0.17

PieroCo, 2015-11-10:
The problem is only on top bar

projectkk2glider, 2015-11-10:
That is strange. If its working on X8R then it should also work on X6R they are almost the same.

indavis, 2015-11-10:
@PieroCo Are you logging the flights? I have two High Precision Varios, and it seems that one of them might be on an older firmware as it occasionally gives me crazy altitude values that throw everything completely off. This sounds exactly like what is going on. When the spurious altitude value comes out, the vario goes nuts because it sees an altitude change from say 300ft up to 16,000ft, and then as it goes back to normal.

I am attaching one of my log files from a Radian glider. I have removed the gps location for my own privacy, but left in the GPS Alt so that you can see it is staying pretty constant. Take a look at row 922 and row 5109. As you can see, for a few log entries after that it has thrown the vario for a real loop. I didn't look down at the display to see if it was causing an issue, but I am guessing the size of the number would cause a problem like is being described above.

Edit: I don't have permission to attach. So I went ahead and posted it on rcgroups in the following message:

http://www.rcgroups.com/forums/showpost.php?p=33159602&postcount=1705

janekx, 2015-11-10:
Please try to turn on the filter on the vario senzor

indavis, 2015-11-10:
@janekx If that was directed at me, I just checked my Radian model, and filter was unchecked. I shall check it and see if that helps. I will also refrain from patching my vario just in case we need to test anything. I have another glider with a newer vario that I don't think I have seen these random high altitude values. Which is what I think the original poster is also seeing.

janekx, 2015-11-10:
I got too bad datas at ASS sensor here https://github.com/opentx/opentx/issues/3043 so try to filter datas to know more

projectkk2glider, 2015-11-10:
Related to this issue:

>Trying to change what is displayed on the Top Bar using the radio doesn't seem to work. Select it move up or down to change and the only options is (---). Now the radio displays nothing in that location

projectkk2glider, 2015-11-10:
@indavis your data suggests that the sensor itself is sending bad values. If you look closely you will see that the vertical speed goes berserk (rapid min max changes) when the reported altitude is way off. But this seems to be a different issue from the original one.

One idea for future, perhaps we should include an option to turn on a glitch filter for each telemetry value.

projectkk2glider, 2015-11-10:
The log file from above in csv format: http://wikisend.com/download/714218/FigNewtons_Radian-2015-10-27.csv

indavis, 2015-11-10:
I completely agree that my sensor is sending the spurious values. But I haven't looked down at the screen to see if it is throwing the display off like the issue we are discussing. What I do notice is that it throws the vario off exactly like the submitter is suggesting. 

"*The number will be 4294967xx, (xx will vari 20-29) and if the vario tone is on, a unusual tone is produced at the same time the 4294967xx is displayed. Sometimes the number continues to be displayed, (constant with xx variable) but vario tone appears to return to normal.*" is exactly how the vario responds when I catch these. I wish I could just go out and test this some more, but I have to fix the firewall on this glider. I was merely trying to give some data and still think the submitter should make sure logging is on and see if it's his sensor spitting out some data that is out of bounds.

PieroCo, 2015-11-10:
my vario not send spurios values

projectkk2glider, 2015-11-10:
I found one issue if Alt is negative the display on the top bar will be a very large number like above.

PieroCo, 2015-11-10:
@projectkk2glider I checked again. The problem occurs only when the altitude is negative. With all receivers including X8R

commit f854442c (2015-11-10): Fixes #3047:
* negative altitude on top bar was displayed wrong (same bug affected vertical speed)
* add ability to select an altitude sensor with ft unit for top bar
* display actual sensor unit for top bar altitude
commit a77ca856 (2015-11-13): Fixes #3047:
* negative altitude on top bar was displayed wrong (same bug affected vertical speed)
* add ability to select an altitude sensor with ft unit for top bar
* display actual sensor unit for top bar altitude
closed 2015-11-10 by projectkk2glider