opentx/opentx #3048: model memo text space

Nihonski, 2015-11-10:
When programming modells in the companion, the length of names is quite limited. Thats ok, I think its the best choice for the small transmitter displays.
However it would be convenient to have some text space in the companion for each model, where I can leave my own memos, what I have done, what I changed, etc. for every specific model. 
This text space does not need to be downloaded into the transmitter, but if so, could be just one additonal screen to be accessed from the model menu. Maybe enough to be seen on the transmitter, but editing only by companion.
Would be " nice to have feature".

kilrah, 2015-11-10:
Impossible with the current file format. There is already an open issue about improving that.

closed 2015-11-10 by kilrah

projectkk2glider, 2015-11-10:
It's #2339