opentx/opentx #3050: Add the warning message to CLI option

projectkk2glider, 2015-11-10:
As per https://github.com/opentx/opentx/issues/3000#issuecomment-152249468 we should add some kind of warning to the `cli` firmware option in Companion to notify/warn users of potential risk.

bsongis, 2015-11-10:
I think we should split the CLI option from the DEBUG option.
In the future the CLI could be used for EEPROM transfer, commands from Companion, switch to bootloader mode etc.

bsongis, 2015-11-10:
Side note: this is what I did in the Horus branch if I am not mistaken

projectkk2glider, 2015-11-11:
>I think we should split the CLI option from the DEBUG option.

Good idea!

Sacre100, 2015-11-12:
> I think we should split the CLI option from the DEBUG option.

Yes ! For (some) end user, CLI is useful to debug LUA script directly on the radio using print(...) , however, if CLI doesn't reduce too much the available memory for LUA.

projectkk2glider, 2015-11-12:
The Lua print() is currently dependent on DEBUG option, so we have to split that one too!

mpaperno, 2018-06-06:
CLI can now be enabled w/out DEBUG (and all associated tracing).

closed 2018-06-06 by mpaperno