opentx/opentx #3056: Companion allow switching between Simulator and model configuration

elecpower, 2015-11-11:
Currently have to close Simulator to review all of a model's configuration. If model configuration open one can see the current tab only.

I suspect and maybe wrongly the current approach is to avoid concurrent changes to a model's settings and they not being immediately reflected in the Simulator and then Issues being raised by users.

If the change is implemented a warning could be added the Simulator window.

With such a great flexible product the easier one can access information the better and only increases the potential for more uptake.

(Taranis X9D Plus - Companion 2.1.5 - Win 10)

kilrah, 2015-11-27:
The reason is technical, the simulator currently works on a "static" copy of the original data making it impossible for changes to be transmitted between the 2.
An entire redesign of companion's data interface would be required.

elecpower, 2015-11-28:
@kilrah seems we are not quite on the same page. So, if I am interpreting your comments correctly, the Simulator should work without change if launched modeless by Companion rather than as a modal "child" window. So simplistically the change is the launch method in Companion of the Simulator since it does not expect any subsequent "communication" with the Simulator. This "separation" would satisfy the suggestion.

kilrah, 2015-11-28:
Hmm but yeah after you'd get the problem mentioned in your original comment that it can be misleading. 

If you want to you can use the standalone simulator to do this, with the extra manual step to open your model in it being the "barrier" that makes sure people who do it know what they're into.

elecpower, 2015-11-28:
Yes but why make it more difficult just to reinforce the settings "disconnect".

Rather than change the Sim window with a warning, as I originally proposed, an alternative approach, which is reasonably common, is the alert message dialog presented after requesting a function with a do not show me again checkbox. Downside is the value has to be saved in the apps settings. Yes original user issue can resurface when they forget they stop the prompt. But there is only so much you can do for the end user.

Anyway I think we can leave the discussion here for now. Thanks for your views and keep up the great work.

kilrah, 2015-11-29:
All it takes to open your file in the standalone simu is to save it in the location it's looking for, it's a couple of clicks.

mpaperno, 2017-01-15:
There is now a UI specify the eeprom file to use when you start the simulator ( #4164).  Does that help/solve this?

I personally think a modeless sim with the warning is a bit awkward, and that the "built-in" simulator is good for doing quick checks of a config.  IMHO it's worth putting more effort into making the standalone simulator more "friendly," eg. by adding a command to reload the current eeprom file (eg. if you make changes in Companion).

mpaperno, 2017-01-15:
Actually the more I think about it, the less I agree with my earlier "awkward" prognosis.  The whole warning thing would be awkward, but if done correctly this could be pretty nice.  You could keep editing the model in Companion and then hit a button to update the sim with new eeprom data.  Perhaps it could even go both ways, with update from sumu radio to Companion possible (eg. if we use a temp file or shared memory to exchange data).  If a temp file, then this could also essentially work by launching the simulator in an entirely separate process (perhaps just the standalone exe).

Also I realize that using the same eeprom file from Companion and the standalone Simu could also get awkward (since they'd potentially overwrite each other).  It mostly works but you have to pay attention.

-Max

closed 2019-11-15 by stale[bot]