opentx/opentx #3057: Companion Simulator replace generic identifiers with user defined names

elecpower, 2015-11-11:
To take advantage of the features of the model configuration how about bringing the user defined names, where defined, to the Simulator.

Outputs tabs substitue CHn
GVars tab substitute GVn and FMn
Switches - though likely not to be practical due to layout crowding

(Tarnanis X9D Plus - Companion 2.1.5 - Win 10)

kilrah, 2015-11-27:
Don't understand, on the outputs the name is on the same line as the rest, on the GVARs tab the GVAR name is also on the line right there... FM no room to replace them.

elecpower, 2015-11-27:
Sorry if not making myself clear so if pictures speak louder than words then...
[image]
[image]
[image]

elecpower, 2015-11-27:
There would be less need for this if #3056 were implemented though I still feel it worthwhile in that it brings all information into the sim.

closed 2020-07-07 by stale[bot]