opentx/opentx #3058: LUA insertInput missing many options

u8mydog, 2015-11-12:
I wrote/modified a lua wizard for helicopters using 2.1.6 and insertInput() is missing many options. I've always been under the opinion that it is better to do curves, expo and dual rates on the input and before the mix. This makes it easier to use CYCx and removes possible interactions in the mixes. I see that insertMix() supports the extra options.

Please make the following options available to insertInput():
inputName (vs line name)
curveType
curveValue
flightModes
carryTrim
 And some documentation to tell me how to set weight, offset and/or curve to use GV.

kilrah, 2015-11-12:
Duplicate of #1647

u8mydog, 2015-11-13:
I don't think it is a dup. #1647 covers insertMix(). And insertMix() is looking pretty good. insertMix() provides access curveType/value, flightmodes, etc.

But I'm asking for access to these parameters for insertInput().

I'm trying to put together an environment to compile opentx & companion. I'm a bit more of a hack but I could probably make the changes by using insertMix() as a guide.

kilrah, 2015-11-13:
Well it's related, #1647 is basically a blanket issue to add lots of useful things to the lua API... Whenever it's completed both the mixer and inputs should be taken care of. You can jsut add a comment on the other one.

closed 2015-11-13 by kilrah