opentx/opentx #3062: Close any Lua open files when radio powers off

projectkk2glider, 2015-11-13:
A script might have a file open (by using io library). That file should be closed before the radio powers off.

projectkk2glider, 2015-11-13:
@bsongis I think you *optimized* io library a bit too much :grin:  The metatable for file objects `flib[]` should be registered with at least one method `__gc` which will close the open file in such cases.

bsongis, 2015-11-14:
That's very possible I forgot it!

indavis, 2015-11-15:
Just wanted to say Thanks for looking at this. The file io is a game changer as we can write our own log files. Until this is fixed, my workaround is open file, write data, close file. Probably more file io than necessary, but seems to be keeping my files safe and I haven't seen any speed issues (at least with my sd card).

PR #3097 (2015-11-24): Projectkk2glider/lua additions
Closes #3095, closes #3062, closes #2921.

**WARNING**: none of this was tested on the radio! I only did the test in the `simu` and there everything worked (I only tried `string.format()` for string library).
commit 25ef680b (2015-11-24): Fixes #3062: Close any Lua open files when radio powers off
PR #3105 (2015-11-25): Lua addition ported from master to next
Git commit ID added to stamp
Fixes #2921: getValue() now also returns pilot position
Fixes #3062: Close any Lua open files when radio powers off
Fixes #3095: Lua string library added (only 6k bytes of flash used)
Re #3095: luaGetVersion() also returns version as numbers
(Ported from master 7ef790d3..4d0bd14ad and squashed into one commit)

Closes  #3104

All commits from #3097 squashed into one and applied to the next branch. 

I've also added back `opent-stamp` Makefile target that I foolishly removed in 32d0e3890af7bc1c2fb23e48a7a4ddf68400d655 (I thoug [... cut]
commit 7538fcfb (2015-11-25): Git commit ID added to stamp
Fixes #2921: getValue() now also returns pilot position
Fixes #3062: Close any Lua open files when radio powers off
Fixes #3095: Lua string library added (only 6k bytes of flash used)
Re #3095: luaGetVersion() also returns version as numbers
(Ported from master 7ef790d3..4d0bd14ad and squashed into one commit)
closed 2015-11-24 by bsongis