opentx/opentx #3063: Companion model telemetry sensors layout irregularities

elecpower, 2015-11-13:
Depending on selections for a sensor the line vertical layout can change dramatically.

[image]

Suggestion is to modify the layout to the table format used for Special Functions and Logical Switches. Benefits:
- lessens likelihood of abnormal formatting behaviour
- standardises layout
- allows reduction in tab width (another issue raised on this)
- as more sensors and features are added it should assist in maintaining alignments.

Anyway my five cents worth for consideration when this area is revewed in the future. Keep up the good work.

(Taranis X9D Plus - Companion 2.1.6 - Win 10)

kilrah, 2015-11-14:
Not sure what you mean as it's already done the same way as SFs and LSs, there are just more settings.

bsongis, 2015-11-14:
Someone proposed a dedicated DilalogBox (as done for the Mixer and Inputs)

projectkk2glider, 2015-11-14:
>Someone proposed a dedicated DilalogBox (as done for the Mixer and Inputs)

#3032

elecpower, 2015-11-14:
@kilrah It is two fold. One is an extreme case such as the Cels row where the 'type' and Instance fields are stretched well beyond what is necessary just expand to the 'space' between the sensor name and the 'positive' checkbox. Also changing the precision causes am expansion & contraction of fields. Some fields should be fixed width.

The other is that the common field labels could be moved to column headings. If for a sensor type there is no value required then leave the column blank or grey out the field.

I realise there are more combinations of parameters but SFs and LSs also have varaible number of parameters too.

I too support the idea of a dialogbox which maybe a better way of handling if the current layout gets too cumbersome.

kilrah, 2015-11-14:
> One is an extreme case such as the Cels row where the 'type' and Instance fields are stretched well beyond what is necessary just expand to the 'space' between the sensor name and the 'positive' checkbox.

That's not how it works. The entire sensor panel is one entity and everything is justified to the same width, so when you set one sensor to Cels the panel still needs to keep its width because of the other lines that need that much. So remaining items on that Cels line are stretched, it may look weird but it wouldn't look less weird if they were all stuck on the left with some empty space. If adding space in the middle, where would you add it?

> The other is that the common field labels could be moved to column headings. If for a sensor type there is no value required then leave the column blank or grey out the field.

That would achieve the opposite of what you want, as if each setting had its own column the space of that column would be lost when that parameter isn't used. If doing this the panel would be at least twice as wide, there are MANY more parameters than on LSs and SFs. Currently we reuse the space of unused parameters for those that apply to the current configuration which is how it can be "that narrow".

elecpower, 2015-11-14:
@kilrah Understand where your coming from. No matter which way a row is displayed here is likely to be some degree of weirdness. Therefore taking the inputs/mixes approach may be better. 
Layout is a personal area so sometimes we have to agree to disagree on some aspects. Anyway thought I would provide some feedback and see what others thought.

closed 2019-11-15 by stale[bot]