opentx/opentx #3067: Switch swap function

EmanuelStassar, 2015-11-15:
If for some reason you end up not being happy with the switches you choose for certain functions, you have to go through all pages and change the switch settings. Its all too easy to forget one.

I would be great is there is a simple option where you select 2 switched and they are swapped automatically for all settings.

The same for all other secondary controls like sliders and knobs.

kilrah, 2015-11-15:
But then you might be using the same switch position to trigger 2 things that are not completely related, and this would change something you don't expect to be changed...

EmanuelStassar, 2015-11-15:
Not really as you swap the entire switch for another one. That should always work. Everything related to that switch moves and that is the desired effect.
I found out that when flying its hard for me to use the switches above the right stick. Exchanging them was no fun because you use the switch in a lot of menu items.

projectkk2glider, 2015-11-20:
This is somewhat related to:
*  Easy reorder of Logical Switch and Special Functions in Companion #1536 
*  Possibility of rearranging Flight-Modes in Companion by drag'n drop - autoupdating all dependencies (in mixers etc.) #1891 
*  Companion Logical Switches/Special Functions enhancements #1781

EmanuelStassar, 2015-11-20:
I would like to be able to do this switch swap on the radio itself.

projectkk2glider, 2015-11-20:
A stand-alone Lua script could do that on the radio (without any changes to the OpenTX firmware). Just somebody has to write it :wink:

lshems, 2015-12-14:
There is a good place where this would be feasable: in the general switch configuration for your radio.

We now have the possibility to define them as 2, 3 of toggle, but if I physically interchange the SH with the SE, I still want the switch at the SE position to be referred to as the SE switch, even if it has become a 3state switch. 

In other words:
physical switch position (SA, SB, SC) is linked to hardware switch name (HSA, HSB, HSC) is linked to switch functionality (Toggle, 2 or 3 state).

The switch states have already been prepeared in the new 2.1 switch value assignments!!

lshems, 2019-08-17:
Linked to general radio format

kilrah, 2019-08-17:
Will be handled by new switch assignment system.

closed 2019-08-17 by kilrah