opentx/opentx #3071: ar9x, LOGICAL SWITCH, V2,  impossible decimal value.

hildan76, 2015-11-15:
firmware version 2.1.6

i set

LOGICAL SWITCH L1
Func a<x
V1 A2
V2 7
AND Switch ---
Duration ---
Delay ---

i can't input decial value for V2
A2 set is below

SENSOR4 8.2V
Name A2
Type Custom
ID F103 25
Unit V
Precision 0.0
Ratio 24.9
Offse 0.0
Auto Offset □
Positive □
Filter ▣
 Persist. □
Logs ■

check please.

kilrah, 2015-11-15:
Can't reproduce. You probably forgot to set precision. 

[image]
[image]

hildan76, 2015-11-15:
precision is 0.0

i was wrote
↓

A2 set is below

SENSOR4 8.2V
 Name A2
 Type Custom
 ID F103 25
 Unit V
 Precision 0.0
 Ratio 24.9
 Offse 0.0
 Auto Offset □
 Positive □
 Filter ▣
 Persist. □
 Logs ■

hildan76, 2015-11-15:
my simulator had same problem.

[image]
[image]

and, real video is below.

projectkk2glider, 2015-11-15:
@hildan76 Any chance you have two (or more) sensors named "A2"?

kilrah, 2015-11-15:
I know that is what you wrote... but here it works, both on the simulator and on the radio, and the problem would match with Precision being set to 0.--.

Also, the LS doesn't show the "V" unit, so my guess is you have 2 sensors named "A2" and you chose the wrong one.

hildan76, 2015-11-15:
plese check this video.

https://youtu.be/j0rOGSzUBoY

projectkk2glider, 2015-11-15:
@hildan76 please upload your EEPE somewhere and paste a link here.

hildan76, 2015-11-15:
[image]
fds

projectkk2glider, 2015-11-15:
Confirmed in Companion simulator. There is some kind of problem with the units lookup in the logical switch setup page.

projectkk2glider, 2015-11-15:
The lookup does not take in account that every telemetry value has three values (current, min, max) and just goes to the next value's unit.

PR #3073 (2015-11-15): Fixes #3071: wrong telemetry unit and precision was used for V2 in lo…
…gical switches (radios with ARM CPU other than Taranis)

Closes #3071

Please test before merging.
commit 109faf89 (2015-11-15): Fixes #3071: wrong telemetry unit and precision was used for V2 in logical switches (radios with ARM CPU other than Taranis)
commit 1ce0b71c (2015-11-15): Merge pull request #3073 from opentx/projectkk2glider/issue_3071

Fixes #3071: wrong telemetry unit and precision was used for V2 in lo…
closed 2015-11-15 by bsongis