opentx/opentx #3074: [Taranis]Timers with THt and THs settings start in different positions of stick

MicheleVilla, 2015-11-15:
Timer 1 is set with THt option and Timer 2 with THs option
when stick of throttle move the two timers start in different position
I think that the threshold should be the same for both timers
Threshold for THs option is very close to the lower position of the throttle stick

projectkk2glider, 2015-11-20:
I see no problem:
* THt is meant to be triggered when the throttle is moved from low position. It is normal that the threshold for this is a little higher  to reduce false start-ups.

* THs in this case the timer speed is proportional to the throttle position. If we want to be accurate the start position needs to be as low as possible.

I would not change the THt threshold, only option is to raise the THs threshold to the same level. but the THs accuracy would suffer.

MicheleVilla, 2015-11-20:
Th% timer is proportional to throttle position
Tht and Ths start when stick move from lower position 
one stop timer if stick go back(measure motor time), other continue to run (flight time)
Is right?
In this case I think the threshold for Tht and Ths should be the same and should be in tht position (little higher)

closed 2019-08-17 by kilrah