opentx/opentx #3082: we made the Korean sounds. upload please.

hildan76, 2015-11-18:
hi.
My friend made the Korean sounds.

[image]

change ext. jpg -> zip.

can you upload at open-tx.org sounds menu?

bsongis, 2015-11-18:
I downloaded it thanks, now I have to add the glue to allow the users to download it from Companion and select it on the radio.

bsongis, 2015-11-18:
Unless somebody wants to take it before me?

bsongis, 2015-11-18:
Does the korean grammar look like english grammar? It would mean we don't need a tts_ko.cpp ?

hildan76, 2015-11-19:
i don't know about tts_ko.cpp
i am not programmer.
i am mechanical engineer and rc mania.  ^^
we made just wav files.
copy to sounds/en and sounds/en/system on sd memory.
really need a tts_ko.cpp???

projectkk2glider, 2015-11-19:
What Bertrand meant is this: a number 75.6 is played differently in different languages. For example in English: seventy five point six, in other languages the order can be different. 

If numbers are played well with your language pack and English language settings that means that we don't need to add a special grammar file for Korean.

hildan76, 2015-11-20:
sorry.
i was think that just need sound files.
but, now i think it is difficulty.
so we more think about it.
i want close now  this issue.
but, we try again later as soon as.
thanks.

closed 2015-11-20 by hildan76