opentx/opentx #3083: Bricked Taranis plus

0xCCD, 2015-11-18:
Hi, it seems like i've bricked my taranis somehow while using companion :/ 

I already did it on the same way while upgrading from 2.0.9 to 2.1.5 and everything was smooth as butter, but the update to 2.1.6 was not. i've used companion 2.1.6 and since my dfu-util does not recognize my taranis i've used the mounted TARANIS drive.

My operating system is ubuntu 15.10

Is there any chance to recover the taranis ? it seems like it does nothing, even the display does not turn on :/

0xCCD, 2015-11-18:
hell yeah ... got it working after removing battery 
thanks to http://open-txu.org/home/special-interests/curing-a-bricked-taranis/

closed 2015-11-18 by 0xCCD

aviogit, 2018-01-22:
Bricked my Taranis Plus with my first firmware upgrade attempt. (Welcome to me!)

I was using Ubuntu 16.04 with both flash and microSD mounted (so no dfu-utils issues) and opentx-companion22 version 2.2.1~ppa01~xenial. The flashing was always good, but afterwards there was no normal startup, it only went in bootloader mode. This for a whole day of attempts.

I tried almost everything that came to my mind - except removing the button battery, but it was in the list of desperate-vodoo things to try. Flash was good, fsck'd microSD, left without battery, tried to flash without battery (impossible in companion, who knows why...), dmesg said that STM32 Bootloader was fine, and so on.

Finally, launching companion from terminal and looking at application logs, I realized that to change the startup screen, companion was actually patching the firmware! I tried to flash the original unpatched firmware with standard startup screen, and suddenly it booted up normally. Weird. Actually, I'm not 101% sure that this was the only cause, because I could bet a dime that I already tried to flash a clean image without a custom splash screen.

For what I can see from console logs, both this program and the website need a good amount of md5summing to really check what is being written to the flash. Also, a page with flash layout would help a lot who is in the bricked purgatory - cause md5sums of downloaded and flashed firmware don't match because of STM32 Bootloader (I think) - never mind the patched firmware with custom image.

3djc, 2018-01-22:
Yes, startup screen change requires reflashing, as this bmp is actually embedded into firmware (it is displayed before/while system is loaded). It has been the case for ages and that has never been an issue. As for your particular case, it is hard to tell without more data.