opentx/opentx #3084: [Taranis B][Special Functions] "Haptic" ,vibrates the wrong number of times ?

483917, 2015-11-18:
Windows 10 pro Swedish
Companion 2.1.6N531 Eng
Taranis B ,opentx-taranis-haptic-timer3-lua-mixersmon-sqt5font-en-2.1.6N530.bin

If i set Haptic = 0 then i get 2 vibrations ,should not vibrate at all ,or ?
If i set Haptic = 1 then i get 3 vibrations  ,should vibrate 1 time
If i set Haptic = 2 then i get 4 vibrations  ,should vibrate 2 times
If i set Haptic = 3 then i get 5 vibrations  ,should vibrate 3 times

Test my HAPTIC.bin model
Switch SA- and SB- , triggers the Haptic function

https://www.dropbox.com/s/quz1cz9c9yil0dw/HAPTIC-2015-11-18.bin?dl=0

projectkk2glider, 2015-11-18:
Looks like a simple 1 off problem. I think special function haptic 0 vibrations would make no sense :)

projectkk2glider, 2015-11-18:
We also need to check how this is represented in Companion!

kilrah, 2015-11-19:
It's meant to be "pattern number 0,1, 2 or 3" ;)

projectkk2glider, 2015-12-26:
Indeed, the value defines a pattern: https://github.com/opentx/opentx-taranis-manual/issues/1

closed 2015-12-26 by projectkk2glider