opentx/opentx #3086: ar9x, no works navigation encoder switch.

hildan76, 2015-11-20:
no option "encoder switch" in companion for arm board.
but, simulator have option "RotEncNavig" in RADIO SETUP.
transmitter no have this option.
finally, no work navigation encoder switch.
check please.

bsongis, 2015-11-20:
Who has the board? I don't!

kilrah, 2015-11-20:
I do, but no encoder.

pierrotm777, 2015-11-22:
See here please !
http://www.rcgroups.com/forums/showthread.php?t=2479552

hildan76, 2015-11-22:
i know about ar9x and encoder.
http://www.thingiverse.com/thing:1021368 <== this thing is my design.

just, opentx not support encoder.
i am talk about that.

kilrah, 2015-11-22:
And what I mean is that I do not have an encoder to do any testing with if we want to try and add it ;)

hildan76, 2015-11-23:
buy it!!! haha
another method,
er9x and sky9x is support encoder.
please, check opened source code of er9x, sky9x.
and apply opentx code.
i think, you already have an code.
you need only check option for compile.
really is it impossible?

closed 2019-11-15 by stale[bot]