opentx/opentx #3089: "Entries" menu don't work

JL3R, 2015-11-21:
I am under X9D Taranis 217 and EEPROM Version 2.1.6 openTX
The menu " entries " does not allow me to adjust the dual rates nor the expo.

kilrah, 2015-11-21:
Makes no sense...
You're likely using it wrong. See doc here:
http://open-txu.org/home/undergraduate-courses/basic-airplane-set-up-with-taranis-2/heli-flight-modes-and-inputs-513/

and please give more details if there is really something not working as intended.

closed 2015-11-21 by kilrah

JL3R, 2015-11-21:
thank you for your help, but with 2;1;6 version it done'nt work !

try it plz and confirm me.

projectkk2glider, 2015-11-21:
I see no problem. @JL3R where **exactly** do you have a problem. Describe in more precisely.

kilrah, 2015-11-21:
Thousands of people are using it without issue. 
My guess is you didn't understand the processing order of input lines. 

Please continue the discussion on your post on the French forum, will be easier for you to understand there.