opentx/opentx #3095: Lua getDateTime() returns integer values. I am suggesting an enhancement.

indavis, 2015-11-22:
When we use the getDateTime() routine in lua scripts, the table that is created is integer values. The means that if we are putting together a string like YYYYMMDD, there are no leading zeros on values below 10. January 1, 2015 would come together as 201511. The question is, do we want to fix this using lua, which I have a routine written for, or do we want to fix this in the opentx model code. I am have written and tested the following code in radio/src/lua/api_general.cpp. 

=================================
static void luaPushDateTimeStr(lua_State *L, uint32_t year, uint32_t mon, uint32_t day,
                            uint32_t hour, uint32_t min, uint32_t sec)
{
  char stryear[5];
  char strmon[3];
  char strday[3];
  char strhour[3];
  char strmin[3];
  char strsec[3];
  sprintf(stryear, "%lu", year);
  if (mon<10) sprintf(strmon, "0%lu", mon); else 	sprintf(strmon, "%lu", mon);
  if (day<10) sprintf(strday, "0%lu", day); else sprintf(strday, "%lu", day);
  if (hour<10) sprintf(strhour, "0%lu", hour); else sprintf(strhour, "%lu", hour);
  if (min<10) sprintf(strmin, "0%lu", min); else sprintf(strmin, "%lu", min);
  if (sec<10) sprintf(strsec, "0%lu", sec); else sprintf(strsec, "%lu", sec);
  lua_newtable(L);
  lua_pushtablestring(L, "year", stryear);
  lua_pushtablestring(L, "mon", strmon);
  lua_pushtablestring(L, "day", strday);
  lua_pushtablestring(L, "hour", strhour);
  lua_pushtablestring(L, "min", strmin);
  lua_pushtablestring(L, "sec", strsec);
}

/*luadoc
@function getDateTimeStr()

Returns current system date and time that is kept by the RTC unit as a string with leading zeros

@retval table current date and time, table elements:
 * `year` year
 * `mon` month
 * `day` day of month
 * `hour` hours
 * `min` minutes
 * `sec` seconds
*/
static int luaGetDateTimeStr(lua_State *L)
{
  struct gtm utm;
  gettime(&utm);
  luaPushDateTimeStr(L, utm.tm_year + 1900, utm.tm_mon + 1, utm.tm_mday, utm.tm_hour, utm.tm_min, utm.tm_sec);
  return 1;
}
=================================

As a secondary issue, I am compiling this on a 32 bit system:
isaac@isaac-ThinkPad-T61:~/myopentx/opentx/companion/lbuild$ uname -m
i686
isaac@isaac-ThinkPad-T61:~/myopentx/opentx/companion/lbuild$ getconf LONG_BIT
32

If I use %lu in the sprintf command above, I get a warning when I compile Companion:
/home/isaac/myopentx/opentx/companion/src/../../radio/src/lua/api_general.cpp:168:31: warning: format ‘%lu’ expects argument of type ‘long unsigned int’, but argument 3 has type ‘uint32_t {aka unsigned int}’ [-Wformat]
and no warning when I compile the Model.

If I change it to %u, I get no warning when compiling Companion and this warning when compiling Model (make PCB=TARANIS LUA=YES):
lua/api_general.cpp:168:30: warning: format '%u' expects argument of type 'unsigned int', but argument 3 has type 'uint32_t {aka long unsigned int}' [-Wformat]

Looks like there are two issues here.
1) Do we want to solve the leading zero problem on the getDateTime with a new getDateTimeStr function in the model, or would we rather keep the solution on the lua side. I am honestly not sure which would save more space, but know that we want to keep the solution in the safest most efficient place. If we decide to go with this, I will request a pull with my new code after completing my testing. It's working in the simulator but I want to test it in my radio. 

2) What is the proper sprintf conversion for uint32_t? I can only find %u and %lu, and either one or the other compiles throws a warning.

projectkk2glider, 2015-11-23:
I've taken the liberty to format you post a little.

For your `%lu` problem, have you looked at how it is done in the logs.cpp?

For the new function, I am not sure yet.

projectkk2glider, 2015-11-23:
If we decide to do a C implementation, then this:
```c
if (mon<10) sprintf(strmon, "0%lu", mon); else    sprintf(strmon, "%lu", mon);
```
Should be written as:
```c
sprintf(strmon, "%02lu", mon);
```

projectkk2glider, 2015-11-23:
Another option is to include Lua string library or only a subset of it, for example `string.format()` is pretty universal and should not take much code space.

indavis, 2015-11-23:
Thanks for prettying up my post. I will make sure I use the Markdowns in the future. I like the idea of adding in the Lua string library as that would solve another issue. In Lua, when I getVersion() iit returns for example: 2.1.6. Without the string library I can't take the left 3 chars and determine if it's 2.1 or 2.0. I can only make an exact comparison. 

I changed my test code to sprintf(strmon, "%02lu", mon); and that is working fine. I just still get the warning in Companion if I use %lu" or a warning in Radio if I use "%u". I looked at the log.cpp file, and it is using %d for the date and time logging. So I will try handing the pushDateTimeStr code a different type.

projectkk2glider, 2015-11-23:
@indavis I've made some commit which should address both your problems: version comparison and date formatting. Are you able to test my changes?

indavis, 2015-11-23:
I am out running an errand. I will be able to test later tonight and will reply back. Thanks.

indavis, 2015-11-24:
Looks good. I ran the following Lua:
..c
file=io.open(fname,"w") -- Write out the csv file header.
		local ver, radio, maj, minor, rev = getVersion()
		io.write(file, "Version="..ver.."\nRadio Version="..radio.."\nVersion Major="..maj.."\nVersion Minor="..minor.."\nRevision Number="..rev.."\nVersion Len="..string.len(ver).."\nLeft 2 of ver="..string.sub(ver,1,3).."\nUpper of Radio Version="..string.upper(radio).."\n")
..
And got the following:
..
Version=2.1.6
Radio Version=taranis-simu
Version Major=2
Version Minor=1
Revision Number=6
Version Len=5
Left 2 of ver=2.1
Upper of Radio Version=TARANIS-SIMU
..
The addition of the string library will really help. The getVersion() code will make it so that I can write a script which can determine which version it's running on and use the appropriate routines to get telemetry. 

I compiled the radio with "make PCB=TARANIS LUA=YES" and copied the open-tx.bin to my radio and it just locked up. I had to pull the battery and put it back in to get to the bootloader. So I figure I am doing something wrong with the make for the radio. I have a version 2 Taranis (non-plus) I believe.

bsongis, 2015-11-24:
One comment, whenever it's possible it's better to avoid sprintf (as it does malloc if I am not mistaken).
Perhaps you could have a look at strhelpers.cpp, there are some interesting functions. And obviously you may add new strings helpers there!

bsongis, 2015-11-24:
It would be nice to have the strings library available, we need a small test to know how much RAM it will use. But in all cases it could be an option.

indavis, 2015-11-24:
I have a feeling that with the updates projectkk2glider is making, my code will not be needed/used. According to him above, adding in the string library uses 6k bytes of flash memory.

bsongis, 2015-11-24:
Yes, I don't care of the flash usage, but the RAM is the biggest problem!

projectkk2glider, 2015-11-24:
>Yes, I don't care of the flash usage, but the RAM is the biggest problem!

The string library should not use any RAM, it has no global variables and no meta tables. Only rotable for functions. Checkout 1659d8ba7867374ed1cb1313d85529d8e2bd1f96. But I might be mistaken :smile:

projectkk2glider, 2015-11-24:
>Yes, I don't care of the flash usage, but the RAM is the biggest problem!

The string library should not use any RAM, it has no global variables and no meta tables. Only rotable for functions. Checkout 1659d8ba7867374ed1cb1313d85529d8e2bd1f96. But I might be mistaken :smile: 

@indavis If you have Taranis Plus it will be written on the border of the LCD screen.

projectkk2glider, 2015-11-24:
>One comment, whenever it's possible it's better to avoid sprintf

sprintf() is also used elsewhere in the Lua code. I will have a look at strhelpers.

projectkk2glider, 2015-11-24:
>I compiled the radio with "make PCB=TARANIS LUA=YES" and wrote the open-tx.bin firmware to my radio and it just locked up.

@indavis just tried the commit 69594ba38b39ef5e7a28058f192283cd9819542b on my radio (Taranis) and the radio works, but I haven't tested any of the new Lua functions yet. You ether have Taranis Plus or some other problem. Try:
```
make clean
make PCB=TARANIS PCBREV=REVPLUS LUA=YES
```

indavis, 2015-11-24:
@projectkk2glider Hehe. I know it's not a plus, but I think there was an initial run of Taranis radios and then a second run. Which I guess is totally irrelevant now that I think about it.

projectkk2glider, 2015-11-24:
The script for the test, which runs perfectly on my radio:
```lua
local f
local firstTime = 1

local function init()
  f = io.open("test.txt", "a")
end

local function run(event)
  lcd.clear()
  local verstr, radio, major, minor, revision = getVersion()
  lcd.drawText(0, 0, "Version: "..string.format("%d.%02d.%04d", major, minor, revision), 0)
  lcd.drawText(0, 12, "Running: "..verstr.." on "..radio, 0)

  -- open file and don't close it
  if firstTime == 1 then
    io.write(f, "this is a test @" .. getTime() .. "\n")
    firstTime = 0
  end
  return 0
end

return {  run=run, init=init }
```
GV use: 6532b

Each time this stand-alone script is run new line is appended to a file `test.txt` and the file is not closed. The file gets closed by the garbage collector when we force script exit with EXIT long press.

dsbeach, 2015-11-24:
@indavis - I had issues with my firmware builds until I figured out that I required gvars

try `make PCB=TARANIS PCBREV= HELI=NO GVARS=YES LUA=YES WARNINGS_AS_ERRORS=YES`

indavis, 2015-11-24:
I tried "make PCB=TARANIS PCBREV= HELI=NO GVARS=YES LUA=YES WARNINGS_AS_ERRORS=YES"

```
Creating load file for Flash: opentx.bin
arm-none-eabi-objcopy -O binary  opentx.elf opentx.bin

Size after:
   text	   data	    bss	    dec	    hex	filename
 364656	   2416	  54768	 421840	  66fd0	opentx.elf
-------- end --------
```
So the compile worked just fine. Pointed to it in Companion21 which was also compiled and that also seems to have worked.
```
FlashFirmwareDialog: flashing "/home/isaac/projectkk2glider/opentx-projectkk2glider-lua_additions/radio/src/opentx.bin" 
readFirmware: reading "/media/TARANIS/FIRMWARE.BIN" into "/tmp/20503-flash-check.bin" 
startFlash: checking firmware compatibility between  "/tmp/20503-flash-check.bin" and "/home/isaac/projectkk2glider/opentx-projectkk2glider-lua_additions/radio/src/opentx.bin" 
writeFirmware: writing "/media/TARANIS/FIRMWARE.BIN" from "/home/isaac/projectkk2glider/opentx-projectkk2glider-lua_additions/radio/src/opentx.bin" 
```
Radio locks up after exiting the bootloader. I have to pull the battery to get back to the bootloader. Since I will assume this is an issue with my end, if either one of you want to compile this and send me a firmware I will load it in and give it a test.

dsbeach, 2015-11-24:
Are you flashing with the radio off when plugging in usb?  That will update the bootloader and firmware in one pass, which is my preferred method.

You'll need zadig if you have not done it that way before.

projectkk2glider, 2015-11-24:
@indavis try switching to the master branch. That one should work, if it doesn't then your build procedure is problematic. Are you sure you don't have Taranis plus? Do you have haptic inside?

projectkk2glider, 2015-11-24:
@indavis also make sure that whenever you change the make compile options you have to first run `make clean`! Otherwise the old options might be used.

indavis, 2015-11-24:
@projectkk2glider I am positive it's not a Plus. No haptic. I also made sure to do the make clean prior to building. Let me know if you want me to work on this offline so we don't keep cluttering up this thread as I agree with you, it's 99.99% something with my build. I am building it from this file:

15760894 Nov 23 21:37 opentx-projectkk2glider-lua_additions.zip

projectkk2glider, 2015-11-24:
Lets got to gitter.

projectkk2glider, 2015-11-24:
>One comment, whenever it's possible it's better to avoid sprintf (as it does malloc if I am not mistaken).
Perhaps you could have a look at strhelpers.cpp, there are some interesting functions. And obviously you may add new strings helpers there!

sprintf is used every time Lua needs to convert a number to string:
```c
#define lua_number2str(s,n)	sprintf((s), LUA_NUMBER_FMT, (n))
```

PR #3097 (2015-11-24): Projectkk2glider/lua additions
Closes #3095, closes #3062, closes #2921.

**WARNING**: none of this was tested on the radio! I only did the test in the `simu` and there everything worked (I only tried `string.format()` for string library).
commit 1659d8ba (2015-11-24): Fixes #3095: Lua string library added (only 6k bytes of flash used)
PR #3105 (2015-11-25): Lua addition ported from master to next
Git commit ID added to stamp
Fixes #2921: getValue() now also returns pilot position
Fixes #3062: Close any Lua open files when radio powers off
Fixes #3095: Lua string library added (only 6k bytes of flash used)
Re #3095: luaGetVersion() also returns version as numbers
(Ported from master 7ef790d3..4d0bd14ad and squashed into one commit)

Closes  #3104

All commits from #3097 squashed into one and applied to the next branch. 

I've also added back `opent-stamp` Makefile target that I foolishly removed in 32d0e3890af7bc1c2fb23e48a7a4ddf68400d655 (I thoug [... cut]
commit 7538fcfb (2015-11-25): Git commit ID added to stamp
Fixes #2921: getValue() now also returns pilot position
Fixes #3062: Close any Lua open files when radio powers off
Fixes #3095: Lua string library added (only 6k bytes of flash used)
Re #3095: luaGetVersion() also returns version as numbers
(Ported from master 7ef790d3..4d0bd14ad and squashed into one commit)
closed 2015-11-24 by bsongis