opentx/opentx #3096: Speedup the creation of QTableWidget pages

projectkk2glider, 2015-11-23:
They are used in Oputputs, Logical switches and Special functions since #2861. 

An idea from @griebd https://github.com/opentx/opentx/pull/3091#issuecomment-158797813 is:

>in logsdialog.cpp I tried a lot to improve QTableWidget (logTable) when filling it up. Look at lines 485 forward... 

Not sure it will work here, because each cell has its own widget and layout.

483917, 2015-11-28:
Hello, i have a laptop Windows 10 pro 32bit ,2Gig ram ,Intel 4x1,46Ghz processor
and with Companion 2.1.6N530 "open/edit" a  model takes 1 - 2 Sec 
but with Companion 2.1.6N550 "open/edit" a model takes 30 - 40 Sec 
this QTableWidget seems to bee way to slow to use,
mabye go back to the previous lists and put some auto moving/resizing Labels above the columns ?

kilrah, 2015-11-28:
Hmm there's an issue indeed, also takes about 7 sec to open a model edit window for Taranis Plus on Windows.
Seems the issue is mostly on the LS page as each time I click on that I have a 2-sec freeze while SFs and outputs come up instantly. 
Missing lock?

projectkk2glider, 2015-11-28:
>Missing lock?

I always wondered what was `lock` used for :smile:  Now it makes more sense. This is very possible reason for slowness.

kilrah, 2015-11-28:
When you change the value of an object (whether manually or programmatically) it triggers its valueChanged method or equivalent. Said method calls update(), which changes values of other objects, which triggers their changed methods, which without lock would call update() again... i.e. changing one thing could cause an insane amount of cascaded calls.

projectkk2glider, 2015-11-29:
I've added some timing code and came up with this (on my machine opening empty model):

The overall time was  (first time is time taken from the previous printout or from start, second number in [] means cumulative time):
```
"global" "ModelEdit creation" "0 ms [0 ms]" 
  "ModelEdit" "Outputs" "232 ms [359 ms]" 
  "ModelEdit" "LS" "298 ms [657 ms]" 
  "ModelEdit" "CF" "917 ms [1574 ms]" 
  "ModelEdit" "end" "73 ms [1647 ms]" 
"global" "ModelEdit shown" "20 ms [1668 ms]" 
```

Outputs take 230 ms most of the time is spent in the loop adding elements to the QTableWidget, update() takes only 3ms:
```
"Channels" "header" "2 ms [2 ms]" 
"Channels" "add elements" "225 ms [227 ms]" 
"Channels" "end" "3 ms [230 ms]" 
```

Similar for LS, except here the update()  takes 137ms:
```
"LogicalSwitchesPanel" "header" "2 ms [2 ms]" 
"LogicalSwitchesPanel" "added elements" "151 ms [153 ms]" 
"LogicalSwitchesPanel" "end" "137 ms [290 ms]" 
```

The SF take a lot more time, because it has 64 rows and each combo box has many items:
```
"CustomFunctionsPanel" "get tracks" "1 ms [1 ms]" 
"CustomFunctionsPanel" "get scripts" "0 ms [1 ms]" 
"CustomFunctionsPanel" "add items" "405 ms [406 ms]" 
"CustomFunctionsPanel" "disableMouseScrolling" "1 ms [407 ms]" 
"CustomFunctionsPanel" "update" "445 ms [852 ms]" 
"CustomFunctionsPanel" "resizeColumnsToContents" "61 ms [913 ms]" 
"CustomFunctionsPanel" "end" "0 ms [913 ms]" 
```

The conclusion is:
 * adding items to QTableWidget is slow, no matter what I tried (hide, lock) did not help
 * adding items to QComboBox is slow. For example to add items to SF switch combo-box it takes around 6ms, again tried several things (adding items to the model), but it did not help.

projectkk2glider, 2015-11-29:
So I did some changes to the CF which brought the time from 917 ms to 687 ms, but was unable to do better. Then I came up with the idea to do a lazy population of model tabs when they are first shown. This way the cumulative time is the same, but it is spread out over several user clicks. The implementation can be seen in https://github.com/opentx/opentx/tree/projectkk2glider/issue_3096_populate_on_first_show

kilrah, 2015-11-30:
On what OS are those timings taken?

projectkk2glider, 2015-11-30:
Linux which has 1ms resolution, Windows OTOH has cca 10-15ms resolution for this.

kilrah, 2015-11-30:
Can you commit your timing tests in a branch? It's not so much the resolution that I mind about, but the fact that the times you mention are still "reasonable" on Linux but it seems they can be 10 times longer on Win, which is clearly not OK. Worth seeing if the small improvement you had on linux could by chance have a huge imact on win.

projectkk2glider, 2015-11-30:
My system:
 * CPU i5-3570 CPU @ 3.40GHz
 * 8G RAM

To reproduce my times:
 * debug build of Companion
 * radio profile X9E
 * open new eepe
 * double click on first slot

Result from 6f8c165ce332d1d7f1a0bac38d94a322ba0eadab:
```
"global" "ModelEdit creation" "0 ms [0 ms]" 
"global" "ModelEdit start constructor" "0 ms [0 ms]" 
"ModelEdit" "inputs" "130 ms [130 ms]" 
"ModelEdit" "Mixes" "0 ms [130 ms]" 
"Channels" "header" "1 ms [1 ms]" 
"Channels" "add elements" "227 ms [228 ms]" 
"Channels" "end" "3 ms [231 ms]" 
"ModelEdit" "Outputs" "232 ms [362 ms]" 
"LogicalSwitchesPanel" "header" "1 ms [1 ms]" 
"LogicalSwitchesPanel" "added elements" "151 ms [152 ms]" 
"LogicalSwitchesPanel" "end" "141 ms [293 ms]" 
"ModelEdit" "LS" "301 ms [663 ms]" 
"CustomFunctionsPanel - populate" "get tracks" "1 ms [1 ms]" 
"CustomFunctionsPanel - populate" "get scripts" "0 ms [1 ms]" 
"CustomFunctionsPanel - populate" "add items" "686 ms [687 ms]" 
"CustomFunctionsPanel - populate" "disableMouseScrolling" "1 ms [688 ms]" 
"CustomFunctionsPanel - populate" "update" "2 ms [690 ms]" 
"CustomFunctionsPanel - populate" "resizeColumnsToContents" "3 ms [693 ms]" 
"CustomFunctionsPanel - populate" "end" "0 ms [693 ms]" 
"ModelEdit" "CF" "696 ms [1359 ms]" 
"ModelEdit" "end" "74 ms [1433 ms]" 
"global" "ModelEdit end constructor" "1435 ms [1435 ms]" 
"global" "ModelEdit created" "0 ms [1435 ms]" 
"global" "STARTING MODEL EDIT" "0 ms [1435 ms]" 
"global" "ModelEdit shown" "12 ms [1447 ms]" 
```

kilrah, 2015-11-30:
Ouch.
i7-5970, 4.1GHz, 32GB...

First run:
```
"global" "ModelEdit creation" "0 ms [0 ms]" 
"global" "ModelEdit start constructor" "0 ms [0 ms]" 
"ModelEdit" "inputs" "580 ms [580 ms]" 
"ModelEdit" "Mixes" "3 ms [583 ms]" 
"Channels" "header" "41 ms [41 ms]" 
"Channels" "add elements" "1510 ms [1551 ms]" 
"Channels" "end" "15 ms [1566 ms]" 
"ModelEdit" "Outputs" "1570 ms [2153 ms]" 
"LogicalSwitchesPanel" "header" "10 ms [10 ms]" 
"LogicalSwitchesPanel" "added elements" "804 ms [814 ms]" 
"LogicalSwitchesPanel" "end" "5088 ms [5902 ms]" 
"ModelEdit" "LS" "5945 ms [8098 ms]" 
"CustomFunctionsPanel - populate" "get tracks" "1 ms [1 ms]" 
"CustomFunctionsPanel - populate" "get scripts" "0 ms [1 ms]" 
"CustomFunctionsPanel - populate" "add items" "5148 ms [5149 ms]" 
"CustomFunctionsPanel - populate" "disableMouseScrolling" "5 ms [5154 ms]" 
"CustomFunctionsPanel - populate" "update" "23 ms [5177 ms]" 
"CustomFunctionsPanel - populate" "resizeColumnsToContents" "23 ms [5200 ms]" 
"CustomFunctionsPanel - populate" "end" "0 ms [5200 ms]" 
"ModelEdit" "CF" "5208 ms [13306 ms]" 
"ModelEdit" "end" "529 ms [13835 ms]" 
"global" "ModelEdit end constructor" "13838 ms [13838 ms]" 
"global" "ModelEdit created" "0 ms [13838 ms]" 
"global" "STARTING MODEL EDIT" "0 ms [13838 ms]" 
"global" "ModelEdit shown" "60 ms [13898 ms]" 
```

Second run:
```
"global" "ModelEdit creation" "0 ms [0 ms]" 
"global" "ModelEdit start constructor" "0 ms [0 ms]" 
"ModelEdit" "inputs" "497 ms [497 ms]" 
"ModelEdit" "Mixes" "3 ms [500 ms]" 
"Channels" "header" "7 ms [7 ms]" 
"Channels" "add elements" "1593 ms [1600 ms]" 
"Channels" "end" "16 ms [1616 ms]" 
"ModelEdit" "Outputs" "1619 ms [2119 ms]" 
"LogicalSwitchesPanel" "header" "11 ms [11 ms]" 
"LogicalSwitchesPanel" "added elements" "819 ms [830 ms]" 
"LogicalSwitchesPanel" "end" "5181 ms [6011 ms]" 
"ModelEdit" "LS" "6055 ms [8174 ms]" 
"CustomFunctionsPanel - populate" "get tracks" "1 ms [1 ms]" 
"CustomFunctionsPanel - populate" "get scripts" "0 ms [1 ms]" 
"CustomFunctionsPanel - populate" "add items" "5206 ms [5207 ms]" 
"CustomFunctionsPanel - populate" "disableMouseScrolling" "4 ms [5211 ms]" 
"CustomFunctionsPanel - populate" "update" "23 ms [5234 ms]" 
"CustomFunctionsPanel - populate" "resizeColumnsToContents" "22 ms [5256 ms]" 
"CustomFunctionsPanel - populate" "end" "0 ms [5256 ms]" 
"ModelEdit" "CF" "5264 ms [13438 ms]" 
"ModelEdit" "end" "511 ms [13949 ms]" 
"global" "ModelEdit end constructor" "13951 ms [13951 ms]" 
"global" "ModelEdit created" "0 ms [13951 ms]" 
"global" "STARTING MODEL EDIT" "0 ms [13951 ms]" 
"global" "ModelEdit shown" "46 ms [13997 ms]" 
```

Companion fully uses one core for these 14 secs.

projectkk2glider, 2015-11-30:
My punny comp is 10x faster than yours :grin:

projectkk2glider, 2015-12-01:
Another idea (from Bertrand) is to test the timing with Qt5. I will try to port my changes to qt5 branch.

rdeanchurch, 2015-12-03:
I have a Lenovo ideapad S210, Intel Pentium 2127U @ 1.900GHz 4GB Ram, 64bit OS.
It takes about 6 seconds to open a model on this computer running Win8.1.

projectkk2glider, 2015-12-06:
Qt5 is even slower on my comp:
```
"global" "ModelEdit creation" "0 ms [0 ms]" 
"global" "ModelEdit start constructor" "0 ms [0 ms]" 
"ModelEdit" "inputs" "204 ms [204 ms]" 
"ModelEdit" "Mixes" "0 ms [204 ms]" 
"Channels" "header" "3 ms [3 ms]" 
"Channels" "add elements" "278 ms [281 ms]" 
"Channels" "end" "4 ms [285 ms]" 
"ModelEdit" "Outputs" "286 ms [490 ms]" 
"LogicalSwitchesPanel" "header" "1 ms [1 ms]" 
"LogicalSwitchesPanel" "added elements" "171 ms [172 ms]" 
"LogicalSwitchesPanel" "end" "178 ms [350 ms]" 
"ModelEdit" "LS" "362 ms [852 ms]" 
"CustomFunctionsPanel - populate" "get tracks" "0 ms [0 ms]" 
"CustomFunctionsPanel - populate" "get scripts" "0 ms [0 ms]" 
"CustomFunctionsPanel - populate" "add items" "764 ms [764 ms]" 
"CustomFunctionsPanel - populate" "disableMouseScrolling" "2 ms [766 ms]" 
"CustomFunctionsPanel - populate" "update" "3 ms [769 ms]" 
"CustomFunctionsPanel - populate" "resizeColumnsToContents" "4 ms [773 ms]" 
"CustomFunctionsPanel - populate" "end" "0 ms [773 ms]" 
"ModelEdit" "CF" "775 ms [1627 ms]" 
"ModelEdit" "end" "86 ms [1713 ms]" 
"global" "ModelEdit end constructor" "1716 ms [1716 ms]" 
"global" "ModelEdit created" "0 ms [1716 ms]" 
"global" "STARTING MODEL EDIT" "0 ms [1716 ms]" 
"global" "ModelEdit shown" "12 ms [1728 ms]" 
```
The code is available in branch https://github.com/opentx/opentx/tree/projectkk2glider/issue_3096_timing_qt5, the port of qt5 is not very precise (missing escapes in modelprinter), but the startup should not be affected.

I needed to install:
```
$ sudo apt-get install qtbase5-dev qttools5-dev qttools5-dev-tools qtmultimedia5-dev
```

projectkk2glider, 2015-12-06:
@kilrah can you test the new pull #3131 with the table layout enabled and disabled on Windows?

mpaperno, 2017-03-25:
The way to use QTable* widgets is with a proper [data model](http://doc.qt.io/qt-5/model-view-programming.html).  QTableWidget is not a layout manager, so I'm not surprised performance is horrendous when using it as such. 

There would be some advantages to using Qt data models for storing & displaying OTx model/radio data, display flexibility being a big one.  Working with that kind of structure is also simpler in many ways, eg. no crazy update loops and other "hackery" (like those locks) to make sure the relevant options & choices are displayed.  Also can make it simpler to display subsets of data, which can greatly enhance UI response.  

For example when using a data model with edit/view delegates, it doesn't need to load the full list of available switches for every line of data (into a combo box for example, like now).  It only needs to load the options and draw the combo box when the user actually wants to change the value (click/activate that table cell).  The rest of the time it just needs to show the current value. So the whole thing loads much quicker as a result.

It would take some doing for sure... but the underlying data is already structured (eg. `LogicalSwitchData`) so that's a big plus.

closed 2017-05-12 by mpaperno