opentx/opentx #3098: [Bug][OpenTX 2.1.6] False "Model ID already used" warning

LapinFou, 2015-11-24:
Hi folks,

Here is the steps to reproduce the problem:

1. Create a 1st model with the following RF parameters:
 * Internal Radio System: D16 / Ch: 1-8 / Receiver No. 2
 * External Radio Module: OFF
2. Change the RF parameter:
 * Internal Radio System: OFF
 * External Radio Module: D16 / Ch: 1-8 / Receiver No. 3
3. Create a 2nd model with the following RF parameters:
 * Internal Radio System: D16 / Ch: 1-8 / Receiver No. 2
 * External Radio Module: OFF

At this point you will get the warning **"Model ID already used"**.
Look like the receiver number are check without taking in account if the RF module is effectively ON or OFF.

griebd, 2015-11-24:
Yes, we now :wink:. This one is a duplicate of #2187. Closing this one as there are more info on and discussion going on there.

closed 2015-11-24 by griebd

LapinFou, 2015-11-24:
OK. I missed this one. The issue title was misleading (talking about DSMx).
:smiling_imp:

griebd, 2015-11-24:
Sure, I though so, don't worry!