opentx/opentx #3099: [OpenTX 2.1.6] "No Impulse" on X8R seems to be not working

LapinFou, 2015-11-24:
Hi folks,

In order to create a wireless buddybox (same goal as the "Linker" sold by Aloft), I have done this setup inside the JR module bay of the master radio:
[image]
A X8R binded in D16 & Mode 2 (= no telemetry) → S.Bus to CPPM converter → heartbeat pin
The **master/teacher radio** is configured as followed:
[image]
The **slave/student radio**  is configured as followed:
[image]

This setup is working perfectly. FYI, the master RF is not perturbed anymore by the telemetry feedback of the X8R (binded in mode 2). I had some RSSI warnings with the flying model on the master radio when using a X4R/CPPM setup (probably the telemetry feedback of the X4R was too close of the internal RF).

If I switched off the slave radio, I'm expecting the X8R going to failsafe **no pulses** mode. Thus, the master radio will play the "trainer link lost" voice announcement.
Unfortunately it is not the case. Looks like the X8R is going in **receiver** failsafe mode. Indeed if I chose **receiver** in the slave radio, I will get the same result as **no pulses**.
The XJT and X8R have been flashed with the latest non-EU firmwares.
I'm not sure where the problem is coming from, it can be:
* S.Bus to CPPM converter which doesn't translate correctly the failsafe flag
* The X8R which got a problem (I will try with another one in the incoming days)
* Can be an OpenTX bug, since choosing **no pulses** or **receiver** failsafe mode on the slave radio generate the same result.

kilrah, 2015-11-24:
"No Pulses" only works for the servo outputs, it's irrelevant for the SBUS output. 

It's the SBUS to PPM converter that AFAIK has no other option than to hold positions on failsafe.

closed 2015-11-24 by kilrah

LapinFou, 2015-11-24:
OK. Fine. :smile: 
However there is something weird:
* Changing the failsafe settings will change the behavior when the slave TX is switched off.
* With **No Pulses**, the SBUS to PPM converter doesn't hold the value. Something else is happening.
* When I used the X4R setup (with CPPM software), the **No Pulses** was working fine and I was also getting the voice announcement when losing the link.

So according yours explanations, the SBUS protocol doesn't support the **No Pulse** failsafe mode?

kilrah, 2015-11-24:
> the SBUS protocol doesn't support the No Pulse failsafe mode?

Correct.

LapinFou, 2015-11-24:
Thanks for answering.
It is perfectly clear now.
:smile: