opentx/opentx #3102: Automatic logging of RSSI and Rx battery level on 2.1.x

mpjf01, 2015-11-25:
A friend was recently flying a glider with his new X9E and had lots of RSSI, telemetry lost and RF signal low warnings, without any loss of control apparently. When he looked at the logs after the flight there were no entries for RSSI or Rx battery level.

He is a flier that has not used any other telemetry (anything other than what the Rx naturally supports) in the past, but has used and relies on the Rx battery warnings and the RSSI warnings since his first X9D a couple of years ago. These values were logged on his X9D running 2.0.17. As the warnings are automatically issued, could the current firmware please be altered to automatically log them on the X9E (2.1.4).

projectkk2glider, 2015-11-25:
Duplicate of #194

kilrah, 2015-11-25:
@projectkk2glider #194 refers to radio voltage, this one about RSSI and receiver voltage, so unrelated.

 @mpjf01 RSSI logging is on by default in 2.1 and has always been, the Logs checkbox for RSSI is enabled by default on sensor detection since the first 2.1 release.
https://github.com/opentx/opentx/blob/master/radio/src/telemetry/frsky_sport.cpp#L270

So he must have turned it off or something. 

For A1/RX it is not turned on by default as not all receivers actually provide a value and it could be misleading.

projectkk2glider, 2015-11-25:
Damn, reading too fast, too early in the morning :blush:

projectkk2glider, 2015-11-25:
Perhaps we should change the default for every new sensor, for logging to be enabled?

flarssen, 2015-11-25:
Already suggested, but with possibility to turn it off: #2692

mpjf01, 2015-11-25:
I have just tried this and it seems to me that one has to go through the "discover sensors" routine before RSSI will get logged as well as (the usual) establishing a special function for setting logging going. Even though it might go against the "rules" for dealing with telemetry sensors I believe that it would be useful if RSSI logging was automatic and a default.

When I added RSSI as a sensor after a log file already existed for that day, RSSI didn't get logged until I generated a new logfile by deleting the old. Not sure if so for other sensors. Would this mean that a sensor change during a day would not show up in the logs until the next day?

flarssen, 2015-11-25:
@mpjf01 data will be logged, but don't match the existing file header, see #2542

kilrah, 2015-11-25:
Setting up your sensor chain and discovering sensors is simply the first thing to do when setting up any model where you intend to use any telemetry. Simple usage habit to get.

mpjf01, 2015-11-25:
I understand that that is the way it has been set up. RSSI warnings are given to users without them having to discover anything. RSSI logging requires sensor discovery. In my view it would not be inconsistent for RSSI logging also to be automatic. It would be unsafe to assume that every X9E user is familiar with the changes that 2.1 has brought, particularly as there is no FRSKY supplied paperwork supplied with the radio alerting them to important differences.
There is an argument for logging of more important parameters to be automatic, not subject to the use of a user specified special function. No doubt it has been discussed and ruled out. I am a heavy user of Arducopter so am used to automatic logging and have seen frequent benefits to users and problem resolution there with no downside.

closed 2019-11-15 by stale[bot]