opentx/opentx #3107: Mark telemetry value as critical,  audio volume independent

janekx, 2015-11-25:
I got idea to have ability to assign the telemetry value the critical sign like log and so in new telemetry...  
This  value will have priority and will be heared even when volume is mute. 
This can prevent unwanted mute of volume in new telemetry 
It is enhancement of my previous issue.https://github.com/opentx/opentx/issues/694
And Bertrands https://github.com/opentx/opentx/issues/130

fraternl, 2015-12-01:
At this moment we have a master volume that can be changed. If that master volume is set low or at mute this master volume needs to be turned up, before the value of track can be played. 
All other sounds in the queue will immediately go louder too....
The master volume needs to be controlled and can't be on full all the time having its other mixes to low volume. This would result in a constant noise.....

To implement a persistent sound it needs to turn the volume up, before it plays it.
This could be, like Bertrand suggested, some preset value as this is very personal.
I would add to that a small check in that it should only turn the volume up, never down....

You're suggesting some telemetry value. Do you mean just a value for a certain sensor.
That's quite rigid....

I would suggest 2 new special functions called "Alarm value" and "Alarm track" that will make a sound at the alternate level. These could then get triggered by logical switches.

janekx, 2015-12-01:
Maybe it will be more clear in settings of alarm value to set what value to play in special volume level.

bsongis, 2019-11-15:
I reopen this one, people asked it in my club

closed 2020-08-11 by stale[bot]