opentx/opentx #3108: Sudden "Low memory" alert and FORMAT of eeprom

Hlaalu, 2015-11-25:
Dear all,

I have a TGY 9X equipped with the 'new' Atmega128A, and OpenTX 2.0.17.

While using a simulator (RealFlight G5), I was changing the Mode in the general setting menu of my TX, while messing up a little bit with the channels assignment settings inside the simulator, and I suddenly got the TX rebooting AND formatting the eeprom. Upon new startup, I noticed I only had something like 30 KB of free space, and an 'empty pgbar' on the main screen of OpenTX (which I don't know what it is... when memory is low nothing is displayed like this... it just says: "low memory, press any key to ignore". Is it the icon for 'bad eeprom' or something?).

Switched the TX off and on again, and the memory went back to 3500 or some, the empty pgbar disappeared and I couldn't reproduce the issue any longer.

I lost my models, but that's not a big issue since I have the backup on my PC. The problem is that I don't know what triggered the formatting. What if it happens during flight?

Could it be that the simulator sent some kind of noise through the jack? 

It's kinda similar to this: https://www.youtube.com/watch?v=ZfN1IhWrSdw&feature=youtu.be, but this guy doesn't mention the 'empty pgbar'...

I hope there's something I can do to prevent it happen again.
Thank you indeed.

kilrah, 2015-11-26:
Hi,

If you did operate the horizontal trims you're likely running into the hardware design error of the 9x that is "revealed" when an atmega128 is fitted:
http://openrcforums.com/forum/viewtopic.php?f=95&t=3594

A similar problem has also been reported with the MENU and EXIT keys that by experience seems to relate more to a damaged processor: http://openrcforums.com/forum/viewtopic.php?f=95&t=5249

Hlaalu, 2015-11-26:
Thank you for your reply.
Well, I actually have to say I wasn't even aware of the horizontal trims issue but... YES I HAVE THE ISSUE! I looked to the wiki page whose link's posted in the thread you indicated, and I'll be doing the fix soon.

The problem is that... yesterday, when I had the issue for the first time, I am sure I didn't touch the trims. I just changed the Mode of my transmitter, and not only I had a reboot, but I had an eeprom format. Now there's now way to reproduce the thing using only MENU and EXIT button, no matter how I try. The only thing I get is a sudden abnormal behaviour (and sometimes a reboot) when I hit the right horizontal trim.

I have two 9X, one of which never had any issues... and yes they both have the 128A. I think I am going to do the mod to both anyway...

What still strikes me is that yesterday I didn't touch any trim. Now the things I am curious about now are:

1. Is there the possibility that being connected to the SIM (and so being Trainer 'slave) triggered some unexpected behavior when I changed the mode (and yes I did press EXIT after changing the mode) while operating on SIM channel assignment?

2. Is there a 'shortcut' for eeprom formatting from inside the menu? Something that I might have mistakenly stroke... 

Thank you very much

kilrah, 2015-11-26:
> Is there the possibility that being connected to the SIM (and so being Trainer 'slave) triggered some unexpected behavior when I changed the mode (and yes I did press EXIT after changing the mode) while operating on SIM channel assignment?

Not that I know of, wouldn't see how.

> Is there a 'shortcut' for eeprom formatting from inside the menu? 

No, not on the 9x.

Hlaalu, 2015-11-26:
Thank you again.

Well, now for sure I know I have the horizontal trim issue, that I'm gonna fix over the weekend for both 9Xs, even if one of these hasn't shown any sign of malfunctioning.

What I wonder is that I ALSO have the MENU and EXIT keys (and maybe up, down, left and right?) issue or not.

In the thread you posted the problem was experienced by only one guy, and there are no replies confirming that the 1K0 resistor solution (explained by Mike) solved the problem or not.
Does anybody know something more about this issue?

Once again, I wasn't able to create a EXIT/MENU buttons-related reboot/crash/format of any kind even if I pressed them randomly and quickly...

Thank you again

kilrah, 2015-11-26:
The thing is that the trim issue could possibly damage the processor and/or corrupt memory contents, then causing the MENU/EXIT crash... there have been several issues of just faulty porcessors in the past too, where they'd jsut randomly shutdown and lose all memory contents.

Maybe a reflash i.e. erase/reprogram cycle can help (the guy said he didn't seem to have the issue anymore after a reflash).

Unless you can reproduce the crash reliably I'll close the issue as it's likely not a firmware problem...

closed 2015-11-26 by kilrah