opentx/opentx #3109: [LUA] table.pack(...) is missing in order to use it vararg function

Sacre100, 2015-11-26:
While writing a function to save data on SD card, I discover that writing LUA function with variable argument is not easy for OpenTx. This is due to the absence of table.pack(...)  function. Writing this kind of function is possible using recursive call but with the limitation that nil argument cannot be processed.

Exemple of vararg function : 

    local function my_vararg(...)
       local args = table.pack(...)
       for k, v in pairs(args) do
          -- process the argument
       end
    end

projectkk2glider, 2015-11-26:
The table library is not included, use a normal table to pass arguments.

Sacre100, 2015-11-26:
Solution found : 

   local function my_vararg(...)
      local args = {...}
      for k, v in pairs(args) do
         -- process the argument
      end
   end

I close the issue, sorry.

closed 2015-11-26 by Sacre100