opentx/opentx #3110: [LUA] inconsistent implementation of io.read(...), io.write(...) and io.flush()

Sacre100, 2015-11-26:
Following code succeed in LUA 5.2 using ZeroBrainStudio but fail in OpenTx.

    local f = io.open("my-file.txt")
    local s = f:read(20)  -- or f.read(f, 20) which is the same
    io.close(f)

Following code succeed in OpenTx but fail in LUA 5.2 using ZeroBrainStudio. 

    local f = io.open("my-file.txt")
    local s = io.read(f, 20)
    io.close(f)

In LUA 5.2, io.read(...) is equivalent to io.input():read(···)  (see documentation http://www.lua.org/manual/5.2/manual.html#pdf-io.read ) and not a function to read data from a file.

Same for io.write(...) which is equivalent to io.output():write(···) (see http://www.lua.org/manual/5.2/manual.html#pdf-io.write)

And for io.flush() which is equivalent to io.output():flush() (see http://www.lua.org/manual/5.2/manual.html#pdf-io.flush)

projectkk2glider, 2015-11-26:
Our IO library has limitations, read https://opentx.gitbooks.io/opentx-lua-reference-guide/content/lib/io.html

closed 2015-11-26 by projectkk2glider

Sacre100, 2015-11-26:
It's not a problem of limitation, it's a problem of adherence to LUA 5.2 specification.

Not adhering to LUA spec mean that if you want to reuse existing code which use io library, you have to adapt it for OpenTx and this is not good.

kilrah, 2015-11-26:
If you try to reuse code that uses a library that isn't in OpenTX it won't work either...

Not all libs are included, and similarly when there are multiple ways to do the same thing not all have been included.

Sacre100, 2015-11-26:
I agree with you Andre, if something is not implemented, it's OK but if something is implemented like reading or writing a file, it should be implemented in a way compliant with LUA spec, not in OpenTx specific way.