opentx/opentx #3113: More swedish translation issues

HakanBastedt, 2015-11-27:
We have found some cosmetic issues with the swedish translation in se.h.txt
and also an issue with the labelling in telemetry screen setup which has to be fixed.TR_VTELEMSCREENTYPE for CPU_ARM has four fields, but not in the swedish version, there are only two.

If it is okay with you I can fix this and give it a try with the github branch and all that.

kilrah, 2015-11-27:
Sure!

projectkk2glider, 2015-12-13:
@HakanBastedt any progress?

HakanBastedt, 2015-12-13:
Yes, fur sure. I am ready and even made a pull request.
I messed up the endlines -  maybe you remember this?
I think the changes are in the next branch.
Maybe I should close this issue?

projectkk2glider, 2015-12-13:
I see now, the pull request was missing reference back to this issue. Does the master (aka 2.1) also need this fix or not?

HakanBastedt, 2015-12-13:
Yes, current and next need this.
I see I was targetting "next", but it is an update for the current version as well.

projectkk2glider, 2015-12-13:
Can you make another pull request with your changes against master? Thanks.

projectkk2glider, 2016-01-13:
@HakanBastedt any progress on pull against `master`?

HakanBastedt, 2016-01-14:
Oh, I missed you previous message. I'll do a pull request in a day or two.

closed 2016-01-17 by projectkk2glider