opentx/opentx #3117: Telemetry Ratio showing as Prescision 0, Should be Precision 1

rdeanchurch, 2015-11-29:
Version 2.1.6N551, Nov 29 2015 and previous 2.1.6 nightlies.
When discovering new telemetry sensors such as A2  via an X4r, the ratio show in Companion as 13.0.
Same for discovering an X8RAnalog Sensor.
On the tx the Ratio is 13.2 as it should be. 

Reading in the .eepe does not change the symptom.  The ratio still shows as 13.0

elecpower, 2015-12-01:
I've also seen the issue but not quite the same. 

I am not sure of the steps to reproduce as have not tested combinations yet however my X8R model is currently displaying ratio 13.0 for RxBt. I have been modifying the X8R model so cannot say what the original settings were when first downloaded from the radio.

On the other hand my X4R model is displaying 13.2 for RxBt and A2 (I didn't even think I should even get A2 for an XnR so maybe another bug or maybe a hangon from D series) after discovery and then reading settings from radio.

elecpower, 2015-12-05:
Performed some more tests
Test 1
Telemetry tab - Manually change RxBt or A2 to 13.2
Click any other tab
Click Telemetry tab - values changed to 13.1
Click any other tab
Click Telemetry tab - values changed to 13.0
Click any other tab
Click Telemetry tab - values remain at 13.0
Test 2
Telemetry tab - Manually change RxBt or A2 to 13.3 or 13.4
Click any other tab
Click Telemetry tab - values remain at set value

kilrah, 2015-12-08:
Can't reproduce any of this...

What radio type? Can you post, starting from a blank model, a precise step by step list of actions that can reproduce it?

elecpower, 2015-12-08:
Taranis X9D Plus Companion 2.1.6 general release Ubuntu 14.04 LTS
Checked on Win 10 and does not have same issue as test above. Rechecked Ubuntu and can reproduce again but need some time to build from scratch. Also need to check both OS after rediscovering the sensors to see if initial ratio 13.0 or 13.2.

elecpower, 2015-12-08:
On Ubuntu
Open Companion
Create New Models and Settings File
Double click model 01
Create a basic plane using the wizard
Click the Telemetery tab
Manually enter a BxTx sensor using detected settings from other model with Ratio 13.2
Repeat click tabs back and forth and Ratio decreases by -0.1 twice down to 13.0
Reset to 13.2 and pattern repeats.
Running the test a few times have struck a variation where 13.2 drops to 13.0 in one click away and back.
Setting Ratio to 12.9 or 13.3 neither decrease variants occurs.

projectkk2glider, 2015-12-09:
It looks like a float to integer rounding error. I remember similar problem in some other place, where better rounding was used. Found it in 73cd69c63b1ea5dcb2d40aa7a3a90a8f11f7093d.

projectkk2glider, 2015-12-26:
@rdeanchurch and @elecpower I was unable to reproduce, but I did find some inconsistencies. So the fix will be in tomorrows nightly build. Pleas test and reopen if problem still exists. Thanks.

commit 10d29e93 (2015-12-26): Fixes #3117: Telemetry sensor Ratio parameter rounding problem fixed
commit d734e346 (2015-12-26): Fixes #3117: Telemetry sensor Ratio parameter rounding problem fixed (ported from master)
closed 2015-12-26 by projectkk2glider