opentx/opentx #3122: [Bug][OpenTX 2.1.6][Radio] Extended trim not working

LapinFou, 2015-12-02:
Hi folks,

Yesterday, I was explaining to a friend the difference between extended trim (+/-100% range) and normal trim (+/-25% range). It is working fine under Companion (and also the simulator), but it is not working on my radio. Whatever if the option is checked or not, the maximum trim will not exceed +/-25% of travel range.
FYI, the model I used has been created under OpenTX 2.0.12, then converted by the radio when I switched to OpenTX 2.1.3
I didn't try to create a new model to see if the problem is present. I will do it this evening when back from work.

projectkk2glider, 2015-12-02:
AFAIK you need to release and press the trim again to go over the 25%.

LapinFou, 2015-12-02:
Oups !! Didn't try that. I will try again tonight.

LapinFou, 2015-12-02:
I tried. You're absolutely right. Sorry :disappointed:

closed 2015-12-02 by LapinFou

projectkk2glider, 2015-12-02:
Don't worry, we all have a senior moment like that occasionally :wink:

LapinFou, 2015-12-02:
Senior moment ?? What do you mean ?? Could you, please speak a little bit louder, I don't hear you!
:older_man: :innocent: