opentx/opentx #3123: Companion Telemetry page regression and Simulation failure.

rdeanchurch, 2015-12-03:
In the last few nightly builds, including 556, the changes made to the telemetry tab to make it narrower have disappeared and it is again too wide for the screen to show all of it.

When trying to simulate (from within Companion)  and looking at the telemetry page, after about 10seconds, all telemetry values start to blink.   At this time, the F4 screen will no longer effect the values shown in the simulator's telemetry pages.
Tested on Ubuntu 15.1 today and Win 8 a couple of days ago

I apologize if this is known issue with work in progress.

projectkk2glider, 2015-12-03:
Not a new issue, this was not yet fixed. Duplicate of #2781

projectkk2glider, 2015-12-03:
@rdeanchurch Do you have a slowness issue on Windows when opening settings for a particular model, like described in #3096? Kilrah has about 14 seconds of delay there. What is yours?

projectkk2glider, 2015-12-03:
I have just now seen the second part of your description:

>When trying to simulate (from within Companion) and looking at the telemetry page, after about 10seconds, all telemetry values start to blink. At this time, the F4 screen will no longer effect the values shown in the simulator's telemetry pages.
Tested on Ubuntu 15.10 today and Win 8 today.

This needs to be checked.

rdeanchurch, 2015-12-03:
Replied in 3096 to speed(?;-) issue.
Around 6 seconds.
Dean

projectkk2glider, 2015-12-03:
@rdeanchurch I can't reproduce the telemetry simulator issue here (self compiled latest Companion on Linux Mint). How did you install Companion on Ubuntu? There is no official .deb package available.

rdeanchurch, 2015-12-03:
I installed alien with apt-get.
Then used
  sudo alien -k --scripts companion21-2.1.6556-i686.rpm
to convert to companion21-2.1.6556-i686.deb, then
  sudo dkpg -i companion21-2.1.6556-i686.deb

But the telemetry screen blink of telemetry values is the same in win8.1 and Ubuntu.

I've just started trying to learn Linux/Ubuntu and took advice off the web.  I'm a very old/retired programmer that ended up in management while all the guys that worked for me transitioned to Unix...and I didn't do the transition.

So if you told me I installed wrong, I would definitely believe you.

Dean

Calm down gravity is still working!! ....I just tested it.			 

Date: Thu, 3 Dec 2015 14:10:18 -0800

projectkk2glider, 2015-12-13:
@rdeanchurch can you re-check if the second problem still exists in current nightly builds. You will find that the telemetry simulator has changed (a lot) :wink:

rdeanchurch, 2015-12-13:
Looks beautiful!  
Thank you.
Will test.
Dean

Calm down gravity is still working!! ....I just tested it.			 

Date: Sun, 13 Dec 2015 10:52:05 -0800

rdeanchurch, 2015-12-14:
570 Tested on Ubuntu15.10
The telemetry screens still went into showing sensor values blinking, but the telemetry simulator kept on working showing varying values as I changed things.   And a change in the Telem Sim stopped the blinking.
Some weird things in telem Simu, but I asume it is still a work in progress.  
For instance: A2 can simulate 12.5, ..., 12.9, 13.0 but not 13.1  (the parameter to the right shows 13.0 greyed out.)

Great to have the tabs at the top of model simu.  (especially since on Ubunto pushing just F4 gets me there, but in Win8 I have to push Fn+F4.  Switching back and forth causes some mental confusion. (Easy to cause in 79 year old dogs trying to learn new tricks;-)
But the new telem simu concept is freaking wonderful!

dsbeach, 2015-12-14:
@rdeanchurch "a change in the Telem Sim stopped the blinking" - can you elaborate? Did all active telemetry fields stop showing updates, or just the one whose value you were changing?  If just one, which one, and what value were you supplying?

What you are seeing with A2 is expected behavior.  The ratio (greyed value) is the maximum value specified by your transmitter settings.  If you change the ratio in the transmitter, it maximum allowed value will change (after closing the sim window and re-opening it).

rdeanchurch, 2015-12-14:
Tried with 572 on Win8.1
You wer obviously correct re A2 range, my bad.
I meant if I changed a value in the telemetry simulator, the blinking on the model simulation values stopped blinking. I changed SWR from 30 to 2.
The SWR in the model simulation's telemetry page changed from 30 to 2 and the model simulation's telemetry page values stopped blinking.  I left that page on my computer overnight and this morning the telemetry page of the model simulation was still fine.

OK, you were obviously correct on A2 range, sorry, my bad.

Terminology of the various elements of Companion is fuzzy, at least to me.  I hope what I said is clear.
Maybe pics wil help.  
1.Caught A2 in the act of a blink, invoked F4 and 
2.changed A2 value, blinking stopped.

This was on Win8.1, not Ubuntu.
Dean
[ChangedA2inTelSim.zip](https://github.com/opentx/opentx/files/61494/ChangedA2inTelSim.zip)

dsbeach, 2015-12-14:
Thanks for the images, now I understand.

Blinking telemetry values on a telemetry 'Nums' screen indicates that the data is 'stale'. I'm not sure what the threshold is, but stale data means it has not been received by the transmitter is some time.

The telemetry simulator will only send data for A2 when its value is above zero. So, once you make a change in the sim the value is sent and the blinking on a Nums screen is halted. If you change A2's value in the sim back to zero and wait a few seconds, the blinking will return.

This is how it is supposed to work - closing the problem.

closed 2015-12-14 by dsbeach