opentx/opentx #3126: [enhancement] A play value with an extra numeric which will force rounding and precision.

fraternl, 2015-12-05:
Currently the play value gives a spoken output using also the precision, but sometimes the precision is just too much. I would like to propose a way to flexibly round stuff, so one has control over the precision.

As an example I have the variometer that has a resolution of 10 cm. It's a resolution, but not really a precision as this value can be way off. If my glider is flying at height of 233.3 metres the variometer might report me 217.4 metres. If I would do a "play value" at that moment I think it would say 217 metres.
It's already rounding it, which is good.
But I would prefer even more rounding. I would prefer to hear 220 metres then....

With an extra numeric value I would get control over this.
In combination with the delta-switch this would give much more sane output.
With a LUA-script I'm now anouncing each 10 metres.... Doing this with a set of logical switches including a delta one I could come close, but I probably will not get anouncements like "10, 20, 30, 40, 50 metres" but "10,  21, 29, 40, 51 metres"

With an SF like "SF1  Alt Play Value 10" I could make sure the given value is rounded to 10 metres.
Likewise "SF1 Alt Play Value 2" would only give me even values.
The scale could go from 0.001 to 100....

If that value is omitted it could work like it is now, which is quite good already.

But given that precision is not the same as resolution a way to have more control over what's said by the radio would be welcome to me.

kilrah, 2015-12-06:
Unless I'm missing something if you just set the precision to what you want in the sensor settings you'll get what you set as a voice announcement.

fraternl, 2015-12-06:
Hi Kilrah...
No I want to round independent from the sensor setting.
A lot of extra possibilities then and simplified programming.
I need the rounding to take place when it is played...
For height for instance I could make this 10 above 100 metres and 1 below...
I would never hear 104 metres, but only 100 or 110

kilrah, 2015-12-06:
You can make copies of the sensor with different précisions.

fraternl, 2015-12-06:
Well, that's the thing... LUA is not something you could change at the field. This option would enable to do much more using standard programming and it looks clean.

During flying the focus is on the model and what it's doing. A voice alert is a distraction, but it's also informative. If it tells me the model is flying at 190 metres this is informative, if it says 192 metres it is getting more distractive than informative (it takes longer and I need to round it in my head).
This is just an example... 
The rounding shouldn't be only 0.1, 1, 10, 100  but also 0.1,0.2,0.3,0.4, 0.5,0.6,07,0.8,0.9,1,2,3,4,5,6,7,8,9,10,20,30,40,50,50

I thought an extra numeric value would be a clean way to force this rounding.

kilrah, 2015-12-06:
> The rounding shouldn't be only 0.1, 1, 10, 100 but also 0.1,0.2,0.3,0.4, 0.5,0.6,07,0.8,0.9,1,2,3,4,5,6,7,8,9,10,20,30,40,50,50

There's no way we could cleanly integrate a solution that is so flexible, proving my point for a lua script.

And yes you can do it in lua and change the values at the field in multiple ways. 2 as an example: 
- Use the input to a model script to set the value
- Have a script that serves as the setup UI, save that to the SD card, then grab it back in the "processing" script that actually does the voice calls.

fraternl, 2015-12-07:
To be honest I'm postponing to write LUA-scripts. Like you I'm doing a lot of different stuff and can never find the time to stuff into it.

I'm currently using several logical switches for watching the battery it detects a full pack and then resets consumption.
Not only does it take mash switches, but things become quickly hard to read.
I guess I need to make that step and write it in LUA.

The example I gave with the height has already been implemented in LUA. Just not by me, but was added in an existing script on my suggestion.

projectkk2glider, 2015-12-13:
I also vote for the Lua solution.

closed 2016-11-27 by fraternl