opentx/opentx #3130: Companion 2.1 RPM can't be installed

varesa, 2015-12-06:
It seems that the RPM has been packaged in a way that it tries to own `/lib`

```
[esa@zbook Downloads]$ sudo dnf install ./companion21-2.1.6-i686.rpm 
[sudo] password for esa: 
Last metadata expiration check performed 2:13:57 ago on Sun Dec  6 16:01:56 2015.
Dependencies resolved.
================================================================================
 Package                     Arch    Version              Repository       Size
================================================================================
Installing:
 SDL                         i686    1.2.15-17.fc22       fedora          223 k
 companion21                 i686    2.1.6-1              @commandline    7.6 M
 libqzeitgeist               i686    0.8.0-12.fc22        fedora           78 k
 phonon                      i686    4.8.3-3.fc22         fedora          212 k
 phonon-backend-gstreamer    i686    2:4.8.2-3.fc22       fedora          153 k
 pulseaudio-libs-glib2       i686    6.0-8.fc22           updates          23 k

Transaction Summary
================================================================================
Install  6 Packages

Total size: 8.3 M
Total download size: 689 k
Installed size: 20 M
Is this ok [y/N]: y
Downloading Packages:
(1/5): libqzeitgeist-0.8.0-12.fc22.i686.rpm      77 kB/s |  78 kB     00:01    
(2/5): SDL-1.2.15-17.fc22.i686.rpm              164 kB/s | 223 kB     00:01    
(3/5): phonon-4.8.3-3.fc22.i686.rpm             138 kB/s | 212 kB     00:01    
(4/5): pulseaudio-libs-glib2-6.0-8.fc22.i686.rp  47 kB/s |  23 kB     00:00    
(5/5): phonon-backend-gstreamer-4.8.2-3.fc22.i6 139 kB/s | 153 kB     00:01    
--------------------------------------------------------------------------------
Total                                           145 kB/s | 689 kB     00:04     
Running transaction check
Transaction check succeeded.
Running transaction test
The downloaded packages were saved in cache till the next successful transaction.
You can remove cached packages by executing 'dnf clean packages'.
Error: Transaction check error:
  file /lib from install of companion21-2.1.6-1.i686 conflicts with file from package filesystem-3.2-32.fc22.x86_64

Error Summary
-------------

```

projectkk2glider, 2015-12-13:
Is this something new in 2.1.6 or did it also appear with previous versions (2.1.5, etc..)? Self compiled or official rpm from the download server?

closed 2016-01-03 by bsongis

jimgeorge, 2016-02-14:
The .rpm files available at open-tx.org (2.1.7 and 2.1.6) both report this error. Tested on Fedora 22.

adelton, 2017-08-20:
I have built 2.1.9, 2.2.0, and 2.2.master versions of Companion as Fedora rpms at https://copr.fedorainfracloud.org/coprs/adelton/opentx-companion/ and https://copr.fedorainfracloud.org/coprs/adelton/opentx-companion-2.2/.

The .spec's and .src.rpm's with the patches are included in the build result so it should be easy to see how the build was done.

PR #3186 (2016-01-03): Fixes #3130: remove /lib paths from RPM
I have not tested this! Somebody with Fedora should test this. We might need to add some more paths (like `/usr/local/` etc..).

Closes #3130 

References:
* https://cmake.org/cmake/help/v3.0/module/CPackRPM.html#variable:CPACK_RPM_EXCLUDE_FROM_AUTO_FILELIST_ADDITION
* https://cmake.org/Bug/bug_relationship_graph.php?bug_id=13609
commit ad20b963 (2016-01-03): Fixes #3130: remove /lib paths from RPM
commit d8583906 (2016-01-03): Merge pull request #3186 from opentx/projectkk2glider/issue_3130_rpm_problem_fedora

Fixes #3130: remove /lib paths from RPM
commit c5e0eb84 (2016-01-03): Fixes #3130: remove /lib paths from RPM (ported from master)