opentx/opentx #3134: [Taranis X9e][Feature Request] Customize Top LCD voltage display source

copterfan20001, 2015-12-07:
Hi, it would be nice to have one of the telemetry voltage sources display on the large battery bar and numerical LCD on a Taranis X9e. Not sure if this is possible with this hardware though, but still. And not sure if this is the right place to ask, also. Any thoughts?

kilrah, 2015-12-07:
Kinda duplicate of https://github.com/opentx/opentx/issues/2672

copterfan20001, 2015-12-08:
I have dug through sum code, and found that at least it seems to be possible with the hardware. In [opentx/radio/src/main_arm.cpp] (https://github.com/opentx/opentx/blob/d464420a1a6285987126f08f930a2e484bbbf6e4/radio/src/main_arm.cpp#L255) at line 255 I find a reference to the function that sends the appropriate LCD command to the top LCD.

So, it's "just" making a new configuration variable for each model (I'd suggest, in case a model does not have telemetry), and giving the appropriate number source and format to the LCD function. I don't yet feel comfortable doing this myself, despite a small understanding of *C*. Aaaaaaany takers :?)

closed 2019-11-15 by stale[bot]