opentx/opentx #3135: Companion v2.20 simulator screen to large for normal notebooks

hrenz, 2015-12-07:
Hy,
CompanionTx  V2.20 testing,
Simulatorscreen is to large for 15´ Notebooks
Normal Notebook screen solution is 1280x800

Think you all work with big Monitors 19´ to  24´
with a solution of 1920 x 1020  and better
so you cant see the Problem

At CompanionTx V2.16 simulator screen is ok

hrenz, 2015-12-19:
the simulator window shold not larger then 1000x750pixels

projectkk2glider, 2015-12-30:
Any idea how to solve this? We could:
* reduce the size of *gimbals*
* reduce the size of Horus LCD (have 1-1 mapping), but the resulting image might be too small.
* another option is to separate the bottom part (tabs) into another windows, but that brings another set of problems and does not guarantee that the user will be able to see all parts at the same time.

kilrah, 2015-12-30:
The gimbals are huge so I'd start by reducing that and see how that works, if not enough then have an *option* for 1x scaling for those who don't have choice.

tomkfly, 2016-12-08:
Is there any progress on this?

3djc, 2016-12-08:
The issue is that on the other end of the spektrum, my 12,4" but  2 736 x 1 824 laptop display is very very tiny. So a middle point has to be found

tomkfly, 2016-12-08:
The problem with the buttons off the bottom of the screen is it is unusable.

schwabe, 2016-12-08:
in my test the width of the screen is not the problem, the window will just not drawn fully on the screen. If the height is to small on OS X it will just not show. We could try to programmatically reduce the gimbals under a certain resolution and fail with an error message that the simulator needs at least 800 screen height 

@3djc are you running windows/linux and what QT does your version use? http://doc.qt.io/qt-5/highdpi.html Qt 5.6+ should automatically scale it. Iirc it does on OSX.

3djc, 2016-12-08:
[image]

Remember that the whole screen is 12", so it really tiny

closed 2016-12-24 by kilrah