opentx/opentx #3136: Re: [opentx] OpenTX 2.1.6 / 9XR pro USB serial does not work (#3132)

mikeller, 2015-12-07:
mikeller commented 13 hours ago
I noticed that after updating OpenTX on my 9XR pro, the USB serial connection when connecting while the TX is on and running normally has stopped working.

Plugging in USB while off / in bootloader mode still gets the TX recognised as USB drive.

Unfortunately this also means I can not write model configs from companion21 to my TX.

Operating system: ubuntu 15.10

Expected behaviour: TX recognised as /dev/ttyACM0 when plugged in while running

Experienced behaviour: TX not recognised at all (in 'dmesg' output) when plugged in while running
￼
Owner
kilrah commented 3 hours ago
There has never been a USB serial port on the 9XR Pro, and there wouldn't be anything to do with it. You need to be in bootloader mode to read/write models.
￼kilrah closed this 3 hours ago

huh?

Then can you please explain to me why my 9XR pro was recongised as /dev/ttyACM0 when running in normal mode up to 2.1.5, and now it is not anymore?

Also, if you say I have to be in boot loader mode, how do I make companion21 communicate with the TX when it is recognised as USB storage devices?

kilrah, 2015-12-07:
Please don't make a new issue but comment on the existing one. 
Companion precisely configures the radio by reading/writing to files on the USB storage device, it has never used any kind of serial communication.

AFAIK the only way you'd see a serial port is by connecting SUB with the radio powered off, which invokes the processor's internal bootloader to allow programming firmware with the manufacturer's tool (Atmel SAM-BA). 

That's all completely independent of OpenTX.

mikeller, 2015-12-07:
Sorry for that, I wasn't aware that github allows comments on closed issues.

Yes, I believe that is the serial port I am referring to. See screenshot for confirmation. Until 2.1.5, this was what I used to transfer firmware onto my 9XR pro (and it was detected by the host even when the TX was on in 'normal' mode). I used this way because for whatever reason, mounting the drives as USB disks did not get the TX recognised by companion21.

Since I put 2.1.5 on my TX, /dev/ttyACM0 is not recognised by the host any more, TX on or off, in any mode. Is it possible to get that back?

Also, as I discovered now, the transfer of firmware and configuration while mounted as USB drives seems to work for me now, so I've got a way to upload configs, and the above request is purely a 'nice to have'.

[image]

kilrah, 2015-12-07:
Nothing changed on our side regarding this AFAIK. Again the serial port is a feature of the built-in bootloader, not of OpenTX. I'll reopen and let @bsongis confirm...

mikeller, 2015-12-07:
When building master tonight to check if the logical switch / telemetry unit fix works for me (it does), I noticed that there is a setting in Makefile to choose between mass storage / serial / joystick for USB in 'normal' mode. Will build older version tomorrow to see if that brings back serial.

kilrah, 2015-12-07:
That only applies to the Taranis, not the 9XR-Pro.

projectkk2glider, 2015-12-07:
>That only applies to the Taranis, not the 9XR-Pro. 

Kilrah is right!

mikeller, 2015-12-07:
Looking at the code I have to admit that you are right.

Considering this, I am still puzzled as to why companion let me ever flash firmware onto my 9XR pro by using sam-ba over the USB serial port. I was also experimenting with er9x at the time, so at a pinch it might have been that er9x was running on the TX when this happened, and that it therefore was a one-off.

But nonetheless, it might be a useful feature to add to the radio code for the 9XR pro to make it discoverable as a USB serial device, as companion apparently is capable of interfacing with this.

More out on a limb: Would it be possible to make the 9XR pro emulate a USB joystick, like the taranis does? If so, how much effort would be involved in this?

kilrah, 2015-12-07:
> as companion apparently is capable of interfacing with this.

No, once again it isn't. There is nothing it can do with a serial port on any platform. And again your radio will be detected as a serial port for SAM-BA access if firmware has been erased, and maybe if it's off when connected to the computer.

USB joystick is possible, but needs the USB code to be written.

kilrah, 2015-12-08:
That is all correct. Once again... when the drives are not mounted and you try to write it is considered that your radio might have no firmware at all loaded on it (e.g. after pressing the ERASE button on the radio), and thus companion tries to access the processor's built-in bootloader to write the firmware/eeprom using the external SAM-BA tool as a fallback solution. That built-in bootloader pops up when the processor can't find firmware to boot, reports itself as a serial port, and that's what you configure on the first screenshot.

This is thus ONLY for when there is no firmware on the radio. Once there is there is no more reason to use it, you'll never see that serial port again, and all accesses are done using the USB drives. Obviously the SAM-BA window will pop up if you try to write without having connected the radio as the drives won't be found and it's the default fallback but at this point you're just supposed to realise your mistake, cancel, connect the radio and start again.

closed 2015-12-08 by kilrah

mikeller, 2015-12-08:
Ah, sorry, I misunderstood. When you said 'completely independent of OpenTX' I assumed this covered the radio firmware AND companion.

Do you think it would make sense to put some effort into adding USB support to OpenTX on the 9XR pro, to bring it to the same level as USB support on the Taranis? To me, it would be of particular interest to get joystick support to work, and I'd be happy to sink some time into this.

kilrah, 2015-12-08:
Sure, that would be welcome.