opentx/opentx #3138: 9XR Pro: Inverter on serial input not configured

mikeller, 2015-12-09:
I came across a (quiet weird looking) bug last night:

Set up: 9XR pro with OrangeRX openLRS module, modified to output serial (telemetry) on pin 5 of the module connector.

I had had openTX 2.1.6 running on my TX, with everything (including telemetry) working fine.

I then flashed er9sky (r217) onto the TX to check something out.

When I reflashed openTX, config was restored with everything working, except 'No Data' for telemetry.

After tracing all connections involved, I reflashed er9sky, went into telemetry config, and ticked 'invert com1'. Reflashed openTX, restored config. Now everything including telemetry works just like it had before I had flashed er9sky for the first time.

It looks like openTX just accepts whatever value (on/off) the inverter on the internal serial input was set to by the previous firmware, and runs with it.

As a suggestion, what about adding an additional option 'FrSky D (inverted)' to the telemetry protocol list, and switch the inverter on or off based on the user's choice?

If nobody is working on this already, I am happy to sink some time into it.

kilrah, 2015-12-10:
>It looks like openTX just accepts whatever value (on/off) the inverter on the internal serial input was set to by the previous firmware, and runs with it.

I don't see how that could be possible. So you have a transistor inverter in your mod? i.e. the standard mod to make it actually compliant with smart port?

mikeller, 2015-12-10:
No, I don't, all I've got is a pull down on the connection between the serial Tx and pin 5. The 9XR pro has a configurable inverter on the serial input on pin 5 that can be enabled and disabled in the menu. The code that actually switches it is:

```
void writeMFP()
{
        NVIC_DisableIRQ(TWI0_IRQn) ;
        Rtc_write_pending |= 2 ;
        i2c_check_for_request() ; 
        NVIC_EnableIRQ(TWI0_IRQn) ;
}

void setMFP()
{
        MFPsetting = 0x80 ;
        writeMFP() ;
}

void clearMFP()
{
        MFPsetting = 0 ;
        writeMFP() ;
}
```

It is in `radio/ersky9x/src/sky/sound.cpp`, of all places.

kilrah, 2015-12-10:
OK, that probably never was included in OpenTX. Nobody in the team actually uses a 9XR-Pro and only one person has one, so it doesn't get much love.

>As a suggestion, what about adding an additional option 'FrSky D (inverted)' to the telemetry protocol list, and switch the inverter on or off based on the user's choice?

Sounds like a good way to implement it :)

mikeller, 2015-12-10:
Hehe, interesting.

Sounds like an easy fix too, as OpenTX already has most of the things required in place, probably all that's required is to set `MFPsetting`, and then trigger an `i2cCheck()` with `Rtc_write_pending |= 2` (all defined in `radio/src/targets/sky9x/rtc_driver.cpp`).

Seeing that the coprocessor handles the RTC, am I right to assume that this setting will not be erased by pulling the battery, unless the button cell is pulled as well?

In general, I do have a 9XR pro, and I am using it, and I am happy to help, so let me know if I can be of assistance with testing or suchlike.

bsongis, 2015-12-10:
@mikeller are you sure I didn't port this part? I thought I did! I will add the DSMX telemetry (which uses the non inverted serial protocol), but I need the HK module for that and it's out of stock since many months now. So yes, please go on if you want to go deeper in the code, thanks for your help anyhow!

mikeller, 2015-12-10:
@bsongis: Well possible that you did, it certainly looks more different and more integrated than a copy / paste / cleanup would.
What IDE are you guys using?

kilrah, 2015-12-10:
> What IDE are you guys using?

UltraEdit for me

bsongis, 2015-12-10:
Eclipse for me, but I am trying CLion these days.

mikeller, 2015-12-10:
Sweet. Using vim so far, but eclipse sounds promising, if the code inspection features work reasonably well.

bsongis, 2015-12-14:
This one is fixed, right? I guess there was no reference from the commit, or it would have been automatically closed.

mikeller, 2015-12-14:
Yes. My bad for not adding the reference.

closed 2015-12-14 by mikeller