opentx/opentx #3141: Simulator 2.1.6 adding a Fx key for reloading it

pointu, 2015-12-13:
For a faster testing when making scrips could it be possible to add a Fx key for reloading simulator in place of quiting cliquing retying testing values .... ?

projectkk2glider, 2015-12-13:
Do you mean Standalone Simulator or the radio simulator inside Companion? 

In Companion to "reload" one has to:
* click close button and then simulate button
* or press Alt+F4 and then Enter key

Is that too much? Does it warrant a special key combination?

pointu, 2015-12-13:
I'm not explaining it correctly
I mean simulator inside Companion, but it's the same for standalone, and the goal is to "Restart/Reload" it without quiting it, retyping test values etc etc...
when you are modifiying à script for example, you do that externaly from simulator, and you have to quit, re-enter, re-type values, quit re-enter, re-type values...
Is there a way ?

kilrah, 2015-12-14:
I think what he really wants is a way to reset only the lua environment in the simu.
Someone has already suggested him to use a Special Function with a dummy script linked to the momentary switch as that should do what he wants.

pointu, 2015-12-14:
Thanks kilrah, only yhe lua environment, i haven't think that's possible, but where is that suggestion your talking about ?

kilrah, 2015-12-14:
On the French forum

pointu, 2015-12-14:
it's some workaround and questioning not an answer and again an Fx to reload lua environment should be a great idea for not good developer like me who cycle a hight number of time between sim and script editor...

PR #3174 (2015-12-27): Dsbeach/issue3141 fx key to reload scripts
Fixes #3141 Fx key to reload Lua scripts

Added PF7 - Reload Lua Scripts

Implemented via one liner - luaState = INTERPRETER_RELOAD_PERMANENT_SCRIPTS which forces a re-initialization of the Lua environment.

Very handy for script modification and debugging.
commit f6463b64 (2015-12-27): Fixes #3141 - Fx key to reload scripts

Added PF7 - reload Lua scripts
commit f6808c9a (2015-12-27): Fixes #3141 - Fx key to reload scripts

Added PF7 - reload Lua scripts
commit 3e409a03 (2015-12-27): Next branch fixes #3141 Fx key to reload scripts
closed 2015-12-27 by bsongis