opentx/opentx #3146: OpenTX 2.1 - Fahrenheit temperature conversion error

dsbeach, 2015-12-14:
Configure a telemetry sensor (Tmp1 or Tmp2) to display Fahrenheit values. Using the telemetry simulator a value of 100C will show as 180.32.  If you use the transmitter to select Celsius it will show 100C - then switch back to Fahrenheit and the correct value of 212F is shown.

Similarly using 1C you get 2.12F initially, but switching back and forth results as 33F as expected.

projectkk2glider, 2015-12-26:
Another thing I can not reproduce today! :confused:  @dsbeach can you check again with some recent build. The factor 1.8032 also does not appear anywhere in the code.

dsbeach, 2015-12-26:
Will do.

kilrah, 2015-12-26:
Could be a problem with PREC. Using anything else than PREC0 with °F will break the conversion (because of the offset). The radio will force it to 0 when selecting °F, but it seems companion doesn't, it hides the prec field but omits setting it to 0.

dsbeach, 2015-12-26:
Yes, it looks like an issue with PREC.  When editing the model in companion I changed the unit to 'raw' and the precision showed '2'.  Change the precision to '0', change the unit to degrees F and the problem is gone.

The radio (both physical and in the sim) allows the setting of PREC when the unit is degrees C.  Changing to F then hides the field.

I'm not sure if this needs a fix, but regardless it is a low priority.

commit 12e15650 (2015-12-26): Fixes #3146: fixed Fahrenheit temperature conversion error (if PREC was not 0)
commit 400a6c20 (2015-12-26): Fixes #3146: fixed Fahrenheit temperature conversion error (if PREC was not 0)  (ported from master)
closed 2015-12-26 by projectkk2glider